rhboot/fwupdate #3: help to "unharcode" paths for gnu-efi-lib to make fwupdate build in other distros

jristz, 2015-06-24:
Not sure if this because fwupdate is not taking pkgconfig corretly or because the patch is hardcoded but make call /usr/lib/gnuefi//crt0-efi-ia32.o wich is different in some distros.
In Arch (yeah, I'm trying to  make a MAKEPKG for fwupdate for Arch) the path for crt0-efi-ia32.o in gnu-efi-lib is simply usr/lib/crt0-efi-ia32.o

gnu-efi in arch only have this [setup](https://projects.archlinux.org/svntogit/packages.git/plain/trunk/PKGBUILD?h=packages/gnu-efi-libs) therefor I can build fwupdate (from git) without seed magic, but that is plain wrong, so, some hints help?

jared-dominguez, 2015-06-24:
Take a look at https://github.com/rhinstaller/fwupdate/commit/e36f2f23a1f197babde8caa485228c5cbfbd89a4 and https://github.com/rhinstaller/fwupdate/commit/727c4250bf343072a0912bc3dd77f4ef24103b60

You may also find these helpful:
http://linux.dell.com/cgi-bin/cgit.cgi/debian/fwupdate.git/tree/debian/rules
http://linux.dell.com/cgi-bin/cgit.cgi/debian/fwupdate.git/tree/debian/libfwup0.install
http://linux.dell.com/cgi-bin/cgit.cgi/debian/fwupdate.git/tree/debian/fwupdate.install

mirco, 2015-06-24:
Also take a look at this package I added a couple hours ago:
https://aur4.archlinux.org/packages/fwupdate

jristz, 2015-06-24:
That package was not there when I post this.
let check but I thing all is fixed

jristz, 2015-06-25:
I talk so early
 libfwup.c:13:21: fatal error: efiboot.h: No such file or directory
efivar in arch install [usr/include/efivar/efiboot.h](https://www.archlinux.org/packages/testing/i686/efivar/files/) but look like either the stable and the git can't catch the path correctly

jared-dominguez, 2015-06-25:
On Wed, Jun 24, 2015 at 09:34:45PM -0500,  Pablo Lezaeta Reyes [pˈaβ̞lo lˌe̞θaˈeta rˈejɛ] wrote:
>I talk so early
>libfwup.c:13:21: fatal error: efiboot.h: No such file or directory
>efivar in arch install usr/include/efivar/efiboot.h but look like either the
>stable and the git can't catch the path correctly

With which version of efivar are you building?

jristz, 2015-06-25:
Arch have in [core <https://www.archlinux.org/packages/core/i686/efivar/>]
efivars 0.15.
I not build any efivar, only i'm using the (for now) stock efivar in Arch.

jared-dominguez, 2015-06-25:
You need at least efivar 0.19 as noted in fwupdate.spec.in in the source tree.

*Pablo Lezaeta*

vathpela, 2015-07-14:
Is this issue solved, then?

closed 2015-10-05 by vathpela