rhboot/fwupdate #4: Could not find device handle: not found

superm1, 2015-07-10:
Hi,

I'm testing this on a future platform.  I'm noticing that the boot entry isn't getting created even if the tool successfully exits 0 on Linux.  It doesn't show up in efibootmgr output or in the F12 boot menu on the system.

It does stage efi\$distro\fw\fwupdate-x0XphZ.cap on the disk though.

Ignoring that first problem (can be solved later unless it's related I guess), I tried to launch fwupdate from EFI shell.

Here's the output I got:

fs0:\ > fwupdate.efi
Found update fwupdate-6180aaaa-5529-4bb4-b4fd-65b4f788de5b-0
Could not locate device handle: Not Found.
fwupdate: Could not build update list: Not Found

vathpela, 2015-07-13:
Can you hex dump the variable from Linux? I.e. "hexdump -C /sys/firmware/efi/efivars/fwupdate-6180aaaa-5529-4bb4-b4fd-65b4f788de5b-0-0abba7dc-e516-4167-bbf5-4d9d1c739416"

superm1, 2015-07-13:
Sure.  This is from a different invocation after clearing variables and files (from various debug runs trying to look more into this and #5).

```
00000000  07 00 00 00 07 00 00 00  aa aa 80 61 29 55 bf 4b  |...........a)U.K|
00000010  b4 fd 65 b4 f7 88 de 5b  00 00 00 00 00 00 00 00  |..e....[........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 01 00 00 00  02 01 0c 00 d0 41 03 0a  |.............A..|
00000040  00 00 00 00 01 01 06 00  00 17 03 12 0a 00 00 00  |................|
00000050  00 00 00 00 04 01 2a 00  01 00 00 00 00 08 00 00  |......*.........|
00000060  00 00 00 00 00 a0 0f 00  00 00 00 00 88 ac 64 12  |..............d.|
00000070  b9 3c c2 4c 9f 0b d7 65  09 18 52 64 02 02 04 04  |.<.L...e..Rd....|
00000080  4a 00 5c 00 45 00 46 00  49 00 5c 00 75 00 62 00  |J.\.E.F.I.\.u.b.|
00000090  75 00 6e 00 74 00 75 00  5c 00 66 00 77 00 5c 00  |u.n.t.u.\.f.w.\.|
000000a0  66 00 77 00 75 00 70 00  64 00 61 00 74 00 65 00  |f.w.u.p.d.a.t.e.|
000000b0  2d 00 41 00 78 00 35 00  61 00 6c 00 48 00 2e 00  |-.A.x.5.a.l.H...|
000000c0  63 00 61 00 70 00 00 00  7f ff 04 00              |c.a.p.......|
000000cc
```

```
# ls /boot/efi/efi/ubuntu/fw/*
fwupdate-Ax5alH.cap
```

superm1, 2015-07-14:
I compared the DP's that the firmware was recognizing for devices and the DP that was stored in that fwupdate variable.

Firmware recognized DP's via listing out device handles:
```PciRoot(0x0)/Pci(0x17, 0x0), Sata(0x0, 0xFFFF, 0x0), HD(1,GPT, 1264AC88-3CB9-4CC2-9F0B-D76509185264, 0x800, 0xFA000)```

DP Stored in fwupdate-* variable:
```PciRoot(0),Pci(0x17, 0x0), Sata(0x0, 0x0, 0x0)/HD(Part1, 1264AC88-3CB9-4CC2-9F0B-D76509185264, 0x800, 0xFA000)/\EFI\ubuntu\fw\fwupdate-Ax5alH.cap```

The most obvious difference that stands out is that Sata port multiplier is different for the two.

As an experiment I hardcoded the value used when building the DP in libefivar (I'm version 0.20): 
```
diff --git a/src/linux.c b/src/linux.c
index 1231ecf..6f02149 100644
--- a/src/linux.c
+++ b/src/linux.c
@@ -327,7 +327,7 @@ sysfs_parse_sata(uint8_t *buf, ssize_t size, ssize_t *off,
                                        info->sata_info.ata_devno);
        } else {
                *off = efidp_make_sata(buf, size, info->sata_info.ata_port,
-                                      info->sata_info.ata_pmp,
+                                      65535,
                                       info->sata_info.ata_devno);
        }
        return *off;
```

After making this change and rebooting, the system successfully queued up the update and kicked off the rest of the BIOS handling for the Capsule flash.

I did notice this commit in efivar:
https://github.com/rhinstaller/efivar/commit/27c09d58f3277f3ce877973337b4af5d546ff9ec

That clarifies that the problem is with the port multiplier handling of the DP.

vathpela, 2015-07-14:
Okay, so what does "ls -l /sys/class/block/sda" say? (Note - no trailing / . I want to see what the symlink points at.)  It seems as if either I'm parsing the port multiplier wrong from sysfs, or sysfs and the firmware disagree about how the port multiplier should be displayed. (There could always be some other problem too.)

superm1, 2015-07-14:
```
# ls -l /sys/class/block/sda
lrwxrwxrwx 1 root root 0 Jul 14 18:04 /sys/class/block/sda -> ../../devices/pci0000:00/0000:00:17.0/ata1/host0/target0:0:0/0:0:0:0/block/sda
```

vathpela, 2015-07-14:
Sorry - also "ls -l /sys/class/ata_device" ?

vathpela, 2015-07-14:
Also, just to be sure - is sda actually the device in question?  And if not can you do the previous ls on the one that is?

superm1, 2015-07-14:
Yeah, sda is the device in question.
```
# ls -l /sys/class/ata_device
total 0
lrwxrwxrwx 1 root root 0 Jul 14 17:31 dev1.0 -> ../../devices/pci0000:00/0000:00:17.0/ata1/link1/dev1.0/ata_device/dev1.0
lrwxrwxrwx 1 root root 0 Jul 14 17:31 dev2.0 -> ../../devices/pci0000:00/0000:00:17.0/ata2/link2/dev2.0/ata_device/dev2.0
lrwxrwxrwx 1 root root 0 Jul 14 17:31 dev3.0 -> ../../devices/pci0000:00/0000:00:17.0/ata3/link3/dev3.0/ata_device/dev3.0
lrwxrwxrwx 1 root root 0 Jul 14 17:31 dev4.0 -> ../../devices/pci0000:00/0000:00:17.0/ata4/link4/dev4.0/ata_device/dev4.0
```

I should note this is a notebook.  Do notebook sata controllers normally support PMP?  I'm not familiar with the kernel ATA code at all, but is this possibly a situation of the device not supporting PMP but that not being obvious in sysfs files?  At least one reference in kernel code ```drivers/ata/libata-acpi.c``` will set 0xFFFF when the inline ```sata_pmp_attached``` from ```include/linux/libata.h``` is 0.
I might be barking up the wrong tree there though, I don't know.

vathpela, 2015-07-14:
Oh actually, I've just got that completely wrong in libefivar.

vathpela, 2015-07-14:
Yeah, I didn't realize the EFI API here was 0xffff -> no multiplier.

vathpela, 2015-07-14:
It's a little weird that I don't see this issue on any other machines, actually.  Some of them must treat pmp=0 as also meaning none.

vathpela, 2015-07-14:
Can you please try the patch I just pushed to efivar for this?

superm1, 2015-07-15:
Yep, I can confirm that fixes it.

Thanks,

closed 2015-07-15 by vathpela

jiazhang0, 2015-10-12:
I made a patch to resolve this issue. But the patch is based on older commit. I will rebase it to the latest commit.
```C
--- a/efi/fwupdate.c
+++ b/efi/fwupdate.c
@@ -307,6 +307,83 @@ err:
 }

 static EFI_STATUS
+search_file(EFI_DEVICE_PATH **file_dp, EFI_FILE_HANDLE *fh)
+{
+       EFI_DEVICE_PATH *dp, *parent_dp;
+       EFI_GUID sfsp = SIMPLE_FILE_SYSTEM_PROTOCOL;
+       EFI_GUID dpp = DEVICE_PATH_PROTOCOL;
+       EFI_HANDLE *devices;
+       UINTN i, n_handles, count;
+       EFI_STATUS rc;
+
+       rc = uefi_call_wrapper(BS->LocateHandleBuffer, 5, ByProtocol, &sfsp, NULL,
+                              &n_handles, &devices);
+       if (EFI_ERROR(rc)) {
+               Print(L"Could not find handles.\n");
+               return rc;
+       }
+
+       dp = *file_dp;
+       Print(L"Searched Device Path: %s\n", DevicePathToStr(dp));
+
+       parent_dp = DuplicateDevicePath(dp);
+       if (!parent_dp) {
+               rc = EFI_INVALID_PARAMETER;
+               goto out;
+       }
+
+       dp = parent_dp;
+       count = 0;
+       while (1) {
+               if (IsDevicePathEnd(dp)) {
+                       rc = EFI_INVALID_PARAMETER;
+                       goto out;
+               }
+       
+               if (DevicePathType(dp) == MEDIA_DEVICE_PATH && DevicePathSubType(dp) == MEDIA_FILEPATH_DP)
+                       break;
+
+               dp = NextDevicePathNode(dp);
+               ++count;
+       }
+
+       SetDevicePathEndNode(dp);
+       Print(L"Device Path prepared: %s\n", DevicePathToStr(parent_dp));
+
+       for (i = 0; i < n_handles; i++) {
+               EFI_DEVICE_PATH *path;
+
+               rc = uefi_call_wrapper(BS->HandleProtocol, 3, devices[i], &dpp,
+                                      (void **)&path);
+               if (EFI_ERROR(rc))
+                       continue;
+
+               Print(L"Device Path enumerated: %s\n", DevicePathToStr(path));
+
+               rc = EFI_UNSUPPORTED;
+               while (!IsDevicePathEnd(path)) {
+                       Print(L"Comparing: %s and %s\n", DevicePathToStr(parent_dp), DevicePathToStr(path));
+
+                       if (LibMatchDevicePaths(path, parent_dp) == TRUE) {
+                               *fh = devices[i];
+                               for (i = 0; i < count; i++)
+                                       *file_dp = NextDevicePathNode(*file_dp);
+                               rc = EFI_SUCCESS;
+                               Print(L"Match up! Returning %s\n", DevicePathToStr(*file_dp));
+                               goto out;
+                       } else
+                               Print(L"Doesn't match\n");
+
+                       path = NextDevicePathNode(path);
+               }
+       }
+
+out:
+       uefi_call_wrapper(BS->FreePool, 1, devices);
+       return rc;
+}
+
+static EFI_STATUS
 open_file(EFI_DEVICE_PATH *dp, EFI_FILE_HANDLE *fh)
 {
        EFI_DEVICE_PATH *file_dp = dp;
@@ -319,8 +396,11 @@ open_file(EFI_DEVICE_PATH *dp, EFI_FILE_HANDLE *fh)
        rc = uefi_call_wrapper(BS->LocateDevicePath, 3, &sfsp, &file_dp,
                               &device);
        if (EFI_ERROR(rc)) {
-               Print(L"Could not locate device handle: %r.\n", rc);
-               return rc;
+               rc = search_file(&file_dp, &device);
+               if (EFI_ERROR(rc)) {
+                       Print(L"Could not locate device handle: %r.\n", rc);
+                       return rc;
+               }
        }

        if (DevicePathType(file_dp) != MEDIA_DEVICE_PATH ||
@@ -375,7 +455,7 @@ add_capsule(update_table *update, EFI_CAPSULE_HEADER **capsule_out,

        rc = open_file((EFI_DEVICE_PATH *)update->info->dp, &fh);
        if (EFI_ERROR(rc))
-           return rc;
+               return rc;

        rc = read_file(fh, &fbuf, &fsize);
        if (EFI_ERROR(rc))
```