rhboot/fwupdate #5: When adding entry, should use efi_guid_global, not *guid

superm1, 2015-07-10:
In trying to debug #4, I was walking through the code and found an inconsistency for adding Boot#### variables compared to efibootmgr.

Variables are getting added using the same GUID that corresponds to BootNext rather than the EFI Global GUID.

This patch fixes the behavior (and entries start to show up in efibootmgr):
```
commit e632b671466df50da60213493fd81aebba7f527b
Author: Mario Limonciello <mario_limonciello@dell.com>
Date:   Fri Jul 10 13:52:00 2015 -0500

    When adding a Boot#### entry it should be the global GUID not the GUID
    that was found for the BootNext variable.

diff --git a/linux/libfwup.c b/linux/libfwup.c
index 860505d..0ccd88e 100644
--- a/linux/libfwup.c
+++ b/linux/libfwup.c
@@ -694,7 +694,7 @@ do_next:
                        goto out;
 
                sprintf(boot_next_name, "Boot%04X", boot_next);
-               rc = efi_set_variable(*guid, boot_next_name, opt, opt_size,
+               rc = efi_set_variable(efi_guid_global, boot_next_name, opt, opt_size,
                                      EFI_VARIABLE_NON_VOLATILE |
                                      EFI_VARIABLE_BOOTSERVICE_ACCESS |
                                      EFI_VARIABLE_RUNTIME_ACCESS);

```

superm1, 2015-07-10:
This gets it to show up in efibootmgr, but the firmware still isn't seeing it in the F12 menu.  I haven't figured out what the problem is there still.

vathpela, 2015-07-13:
This seems weird though - Why is it finding a BootNext variable that has some guid other than the global guid?  (Don't get me wrong - that means we should be checking this, and we're not currently.)

vathpela, 2015-07-13:
Oh, nevermind.  Obviously this is the case where we didn't find anything, so it's actually just some random guid we've seen before at that point (uh, hopefully...), and I've just not hit this problem because most of them are efi_global_guid and so I'm getting that by luck alone.

vathpela, 2015-07-13:
This should be fixed in these commits:
ea02cdf Fix another place where we should be using efi_guid_global...
d6ba198 Use the correct GUID when setting Boot#### entries.

Thanks!

closed 2015-07-13 by vathpela

superm1, 2015-07-13:
Thanks.  That second commit fixes it getting launched by firmware automatically on next boot (whether or not it's in the boot menu).  It's not listed in the boot menu since it's not in BootOrder, but I think this is more close to expected behavior because you don't want someone randomly booting into that environment when there isn't a file staged to be launched.

vathpela, 2015-10-05:
Yeah - that's right, we put it in BootNext but not BootOrder.  Though strictly speaking there's nothing guaranteeing BootOrder is what's driving any menus either.  In practice that BootOrder and menu interaction across different vendors varies quite a bit - some care about the "active" flag and some don't, some populate BootOrder with things they want in the menu, etc.  This is part of why we changed BootOrder a lot in UEFI 2.5 :)