rhboot/fwupdate #8: fails to build on arm64 and armhf

superm1, 2015-07-16:
Jared pulled in https://github.com/rhinstaller/fwupdate/commit/8f8dcbc7e05c12fbb0f36d2392ace055905d1e22 into the last upload to Debian that synced to Ubuntu.  Under Ubuntu, this has failed to build on arm64 and armhf with the following errors:

arm64(https://launchpadlibrarian.net/211678696/buildlog_ubuntu-wily-arm64.fwupdate_0.4-3_BUILDING.txt.gz)
```
make[3]: Entering directory '/«PKGBUILDDIR»/efi'
gcc -O0 -g3 -fpic -Wall -fshort-wchar -fno-strict-aliasing -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=c11 -DCONFIG_aarch64 -D__KERNEL__ -I/usr/include/efi/ -I/usr/include/efi/aarch64/ -iquote/«PKGBUILDDIR»/include "-DDEBUGDIR=L\"/\"" -ffreestanding -I/usr/lib/gcc/aarch64-linux-gnu/4.9/include -c -o fakeesrt2.o fakeesrt2.c
ld -nostdlib --warn-common --no-undefined --fatal-warnings -shared -Bsymbolic -L/usr/lib -L/usr/lib --build-id=sha1 /usr/lib/crt0-efi-aarch64.o --defsym=EFI_SUBSYSTEM=0xa -o fakeesrt2.so fakeesrt2.o -lefi -lgnuefi \
	/usr/lib/gcc/aarch64-linux-gnu/4.9/libgcc.a \
	-T elf_aarch64_efi.lds
ld: section .note.gnu.build-id loaded at [0000000000000000,0000000000000023] overlaps section .text loaded at [0000000000000000,000000000000668f]
Makefile:74: recipe for target 'fakeesrt2.so' failed
make[3]: *** [fakeesrt2.so] Error 1
rm fakeesrt2.o
```

armhf (https://launchpadlibrarian.net/211678705/buildlog_ubuntu-wily-armhf.fwupdate_0.4-3_BUILDING.txt.gz)
```
make[3]: Entering directory '/«PKGBUILDDIR»/efi'
gcc -O0 -g3 -fpic -Wall -fshort-wchar -fno-strict-aliasing -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=c11 -DCONFIG_arm -D__KERNEL__ -I/usr/include/efi/ -I/usr/include/efi/arm/ -iquote/«PKGBUILDDIR»/include "-DDEBUGDIR=L\"/\"" -ffreestanding -I/usr/lib/gcc/arm-linux-gnueabihf/4.9/include -c -o fakeesrt2.o fakeesrt2.c
ld -nostdlib --warn-common --no-undefined --fatal-warnings -shared -Bsymbolic -L/usr/lib -L/usr/lib --build-id=sha1 /usr/lib/crt0-efi-arm.o --defsym=EFI_SUBSYSTEM=0xa -o fakeesrt2.so fakeesrt2.o -lefi -lgnuefi \
	/usr/lib/gcc/arm-linux-gnueabihf/4.9/libgcc.a \
	-T elf_arm_efi.lds
ld: section .note.gnu.build-id loaded at [00000000,00000023] overlaps section .text loaded at [00000000,0000647f]
make[3]: *** [fakeesrt2.so] Error 1
Makefile:74: recipe for target 'fakeesrt2.so' failed
rm fakeesrt2.o
```

It built fine on amd64 and i386.

vathpela, 2015-09-09:
Can you check and see if this works with 448aaab43 and 05d715fbf63 ?

superm1, 2015-09-09:
Yes, I can confirm that those two commits fix the issue, both arm64 and armhf pass now.

superm1, 2015-09-14:
I noticed you were committing that (and other stuff) specifically to your personal branch (https://github.com/vathpela/fwupdate).  Are you planning on releasing there instead in the future?

vathpela, 2015-10-05:
No, it's just my development branch so I can have people test things before I push to the official repo :)

closed 2015-10-05 by vathpela