rhboot/fwupdate #14: e3b443d introduces a regression if existing fwupdate status variable

superm1, 2015-10-14:
Commit e3b443d introduces a regression if existing variables are around from a previous fwupdate run.

```
$ ls /sys/firmware/efi/efivars/fwupdate-e0f614ed-fb82-467a-a34e-71172cc07e4d-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
/sys/firmware/efi/efivars/fwupdate-e0f614ed-fb82-467a-a34e-71172cc07e4d-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
$ fwupdate -a `cat /sys/firmware/efi/esrt/entries/entry0/fw_class` firmware_0.3.1.cap    // cause a segment fault
$ rm /sys/firmware/efi/efivars/fwupdate-e0f614ed-fb82-467a-a34e-71172cc07e4d-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
$ fwupdate -a `cat /sys/firmware/efi/esrt/entries/entry0/fw_class` firmware_0.3.1.cap    // successful
```

hughsie, 2015-10-14:
It's probably my fault; could you grab a backtrace please.

superm1, 2015-10-14:
WR will need to comment on the backtrace.  I'll ask them to.

I was trying to reproduce this myself, but was having other troubles.
Looking through the code, I noticed the calls to ```open(``` are called with ```O_CREAT``` but don't specify the necessary third argument ```mode``` for when ```open``` is called with  ```O_CREAT``` in ```flags```

reference: http://man7.org/linux/man-pages/man2/open.2.html

This might not be the root of the problem, but I think that should be fixed too.

jiazhang0, 2015-10-15:
This issue is fixed by PR: https://github.com/rhinstaller/fwupdate/pull/17

closed 2015-10-15 by vathpela