rhboot/fwupdate #15: Invalid device path for SSD on Dell SLA

jiazhang0, 2015-10-15:
The generated fwupdate status variable contains an invalid device path on my DELL SLA board. On this board, an USB key (/dev/sdb) and SSD(/dev/sda) are attached. Two drives all uses the first partition as ESP (MBR scheme not GPT). See the infos below.

```
# fdisk -l /dev/sda
Disk /dev/sda: 29.8 GiB, 32017047552 bytes, 62533296 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x5078fe0f

Device    Boot Start       End Blocks  Id System
/dev/sda1         63    273104 136521   e W95 FAT16 (LBA)

# fdisk -l /dev/sdb
Disk /dev/sdb: 7.5 GiB, 8004304896 bytes, 15633408 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x59c88c89

Device    Boot     Start       End  Blocks  Id System
/dev/sdb1           2048    264191  131072   c W95 FAT32 (LBA)
/dev/sdb2         264192   6555647 3145728  83 Linux
/dev/sdb3        6555648  12847103 3145728  83 Linux
/dev/sdb4       12847104  15633407 1393152  83 Linux

# mount | grep sd
/dev/sdb2 on / type ext3 (rw,relatime,errors=continue,user_xattr,barrier=1,data=ordered)
/dev/sdb4 on /media/sdb4 type ext3 (rw,relatime,sync,errors=continue,user_xattr,barrier=1,data=ordered)
/dev/sdb3 on /media/sdb3 type ext3 (rw,relatime,sync,errors=continue,user_xattr,barrier=1,data=ordered)
/dev/sda1 on /boot/efi type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
/dev/sdb1 on /mnt type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)

# rm -f /sys/firmware/efi/efivars/BootNext* /sys/firmware/efi/efivars/fwupdate* /sys/firmware/efi/efivars/Boot0000*
# fwupdate -a `cat /sys/firmware/efi/esrt/entries/entry0/fw_class` ~/update_0.3.2.cap
00000000  07 00 00 00 07 00 00 00  ed 14 f6 e0 82 fb 7a 46  |..............zF|
00000010  a3 4e 71 17 2c c0 7e 4d  00 00 00 00 00 00 00 00  |.Nq.,.~M........|
00000020  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
00000030  00 00 00 00 01 00 00 00  02 01 0c 00 d0 41 03 0a  |.............A..|
00000040  00 00 00 00 01 01 06 00  00 13 03 12 0a 00 00 00  |................|
00000050  ff ff 00 00 04 01 2a 00  01 00 00 00 00 20 00 00  |......*...... ..|
00000060  00 00 00 00 00 20 00 00  00 00 00 00 6a a5 3a ce  |..... ......j.:.|
00000070  5b 51 37 49 a1 f2 b5 a1  a8 e6 f4 74 02 02 04 04  |[Q7I.......t....|
00000080  46 00 5c 00 45 00 46 00  49 00 5c 00 64 00 65 00  |F.\.E.F.I.\.d.e.|
00000090  6c 00 6c 00 5c 00 66 00  77 00 5c 00 66 00 77 00  |l.l.\.f.w.\.f.w.|
000000a0  75 00 70 00 64 00 61 00  74 00 65 00 2d 00 64 00  |u.p.d.a.t.e.-.d.|
000000b0  38 00 42 00 34 00 5a 00  63 00 2e 00 63 00 61 00  |8.B.4.Z.c...c.a.|
000000c0  70 00 00 00 7f ff 04 00                           |p.......|
000000c8

# # hexdump -C BootNext*
00000000  07 00 00 00 00 00                                 |......|
00000006

# hexdump -C Boot0000*
00000000  07 00 00 00 01 00 00 00  60 00 4c 00 69 00 6e 00  |........`.L.i.n.|
00000010  75 00 78 00 20 00 46 00  69 00 72 00 6d 00 77 00  |u.x. .F.i.r.m.w.|
00000020  61 00 72 00 65 00 20 00  55 00 70 00 64 00 61 00  |a.r.e. .U.p.d.a.|
00000030  74 00 65 00 72 00 00 00  04 01 2a 00 01 00 00 00  |t.e.r.....*.....|
00000040  00 20 00 00 00 00 00 00  00 20 00 00 00 00 00 00  |. ....... ......|
00000050  6a a5 3a ce 5b 51 37 49  a1 f2 b5 a1 a8 e6 f4 74  |j.:.[Q7I.......t|
00000060  02 02 04 04 32 00 5c 00  45 00 46 00 49 00 5c 00  |....2.\.E.F.I.\.|
00000070  64 00 65 00 6c 00 6c 00  5c 00 66 00 77 00 75 00  |d.e.l.l.\.f.w.u.|
00000080  70 00 64 00 61 00 74 00  65 00 2e 00 65 00 66 00  |p.d.a.t.e...e.f.|
00000090  69 00 00 00 7f ff 04 00                           |i.......|
00000098

# ls /dev/disk/by-uuid/ -l
[... 149 lines cut ...]
-               Print(L"%a:%a():%d: Could not locate device handle: %r\n",
-                             __FILE__, __func__, __LINE__, rc);
-               return rc;
+               rc = search_file(&file_dp, &device);
+               if (EFI_ERROR(rc)) {
+                       Print(L"%a:%a():%d: Could not locate device handle: %r\n",
+                                     __FILE__, __func__, __LINE__, rc);
+                       return rc;
+               }
        }

        if (DevicePathType(file_dp) != MEDIA_DEVICE_PATH ||
@@ -752,6 +753,11 @@ efi_main(EFI_HANDLE image, EFI_SYSTEM_TABLE *systab)
                return rc;
        }

+       if (debugging) {
+               Print(L"Applying the capsules ...\n");
+               uefi_call_wrapper(BS->Stall, 1, 10000000);
+       }
+
        /*
         * Step 4: apply the capsules.
         */
```

jiazhang0, 2015-10-17:
After a re-creation of GPT on SSD, the first issue is gone. Maybe the relict of GPT is wrongly scanned by efivar and then reports to fwupdate.

jiazhang0, 2015-10-21:
This issue should be closed and move to efivar. The root cause is that efivar will try to search the backup GPT Partition Entry Array and header to get the UUID of partition even they are already corrupted. For example, the end user runs parted to create a ESP with GPT scheme but not rm it. Then running dd to clean the primary GPT Partition Entry Array and header. At this point, Linux kernel cannot recognize current disk with valid GPT scheme and thus cannot find out the ESP on disk.  efivar can still find the UUID of partition from the backup GPT Partition Entry Array and thus generate the wrong device path for status variable and Boot####. efivar should add the sanity check to confirm whether or not a valid GPT exists.

superm1, 2015-10-21:
@jiazhang0 Does this mean that the issue can also be avoided by running partprobe to make the kernel re-read partition tables after dd'ing the image?

jiazhang0, 2015-10-23:
@superm1 partprobe doesn't halp. efivar tries the best to search primary and backup GPT partition entry array to find the valid partition entry info. Actually, the protective MBR has been corrputed. efivar should add a check to make sure the protective MBR is valid then make furthter check to validate GPT.

jiazhang0, 2015-10-25:
Reopen this issue because we have a fix now for fwupdate.

superm1, 2015-10-26:
@jiazhang0 can't this be closed since #23 was closed then?  Or do you think there should be a second fix for this?

jiazhang0, 2015-10-27:
@superm1 Yes. Actually there is another issue as I mentioned initially (Need a patch applied to fwupdate.efi). I will submit another fix later. Close this issue.

closed 2015-10-27 by jiazhang0