rhboot/fwupdate #41: Compiling with -02 causes failure

superm1, 2016-02-29:
This is on Ubuntu 16.04 with it's current default compile flags and master.  `-O2` is used instead of `-O0`
```
gcc -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -fpic -Werror -Wall -Wextra -fshort-wchar -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/home/test/fwupdate/include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/usr/lib/gcc/x86_64-linux-gnu/5/include -c -o fwupdate.o fwupdate.c
fwupdate.c: In function ‘open_file’:
fwupdate.c:584:2: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  UINTN sz = *(UINT16 *)file_dp->Length - 4;
  ^
cc1: all warnings being treated as errors
Makefile:90: recipe for target 'fwupdate.o' failed
```

These were masked previously but 84dae1757fc55c1f20fcc1e4e5e46e18196a03d5 added `-Werror` so this causes a failure now.

vathpela, 2016-03-23:
This should be fixed now, yes?

superm1, 2016-03-23:
No, I still see this:

```
$ :~/fwupdate/efi# gcc -O2 -g3 -fpic -Werror -Wall -Wextra -fshort-wchar -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/root/fwupdate/efi/../include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/usr/lib/gcc/x86_64-linux-gnu/5/include -c -o fwupdate.o fwupdate.c
fwupdate.c: In function ‘open_file’:
fwupdate.c:600:2: error: dereferencing type-punned pointer will break strict-aliasing rules [-Werror=strict-aliasing]
  UINTN sz = *(UINT16 *)file_dp->Length - 4;
  ^
cc1: all warnings being treated as errors
```

Retrying with `-O0` lets it through with no problem

closed 2016-08-11 by vathpela