rhboot/fwupdate #58: Alternative EFI locations

GTB3NW, 2016-12-10:
Hi I'm currently struggling to use fwupdate due to the location of my efi partition. I have it mounted to /esp as FAT32 and /boot is a bind mount to /esp/EFI/arch. Due to it being fat32 I can't do a symlink to support fwupdate's expected folder structure. The exact process for bind mounts is shown here: https://wiki.archlinux.org/index.php/EFI_System_Partition#Using_bind_mount

and isn't too uncommon in the arch community. Would you please look at supporting alternative mount pounts.

Thanks! :)

vathpela, 2017-03-06:
I would happily take a patch to make this a build option :)

closed 2017-08-14 by vathpela

ArchangeGabriel, 2017-09-03:
Making this a compile time option won’t help a lot… Arch users will still face an issue if they don’t use `/boot/efi/EFI` because that’s the way I’ll compile it.

tchernomax, 2017-09-04:
@ArchangeGabriel I also am on Archlinux and suffer from this bug ; I just pushed PR #78 to fix this issue.
On my computer I can't update the firmware since it's already the latest version but the patch seems to work (I check with strace).