rhboot/fwupdate #62: Bios update fails on Dell 9350 (developper edition)

fredoche, 2017-03-07:
Hi,
Gnome-softwate offers me to update my bios to the latest version (0.1.4.10 -> 0.1.4.13)
However, nothing happens when I click "install": the button disappear and comes back a few seconds later. Nothing happens on reboot either.

I'm using a freshly installed fedora 25.

How can I help troubleshooting this issue?

Thanks

superm1, 2017-03-08:
Hi,

Can you please try to install from the command line and see if any errors are happening?
`
# fwupdmgr update

`If no errors are displayed, check and see if a boot entry was created.

`# sudo efibootmgr -v
`
There should be an entry for Linux-Firmware-Updater.

fredoche, 2017-03-08:
Hi,

I did that and it worked flawlessly. My bios, tpm etc, are all updated. Thanks!
However, I checked my command history and I tried this command (`fwupdmgr update`) before writing this ticket as well, and it did nothing (though I'm can't be specific on whether I did something bad somehow, as I've tried every fwupdmgr command before opening that issue).
There is still the issue with gnome-software not reporting TPM or failing to trigger the bios download or update.
My problem is solved but I may still run commands to help you with the rest of that issue?

superm1, 2017-03-09:
Well that's really odd if it "just started working".  
Both TPM and FW updated at the same time?  I wonder if that's possibly related (it's new that the updates for TPM are available).

Can you file an issue with the fuwpd project to investigate further?  I don't think the issue is with the fwupdate project and this issue can close.

fredoche, 2017-03-31:
Does it matter if the Linux-Firmware-Updater entry comes first or second? I'm having the same issue (theres been another tpm update) and this time fwupdmgr update says that there is no update, even though efibootmgr -v shows a specific entry for firmware update.
```

```
> 
> BootCurrent: 0000
> Timeout: 0 seconds
> BootOrder: 0000,0002
> Boot0000* Fedora	HD(1,GPT,597ea7fd-7be1-488c-81e3-2a4a7b6c779d,0x800,0x64000)/File(\EFI\fedora\shim.efi)
> Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,597ea7fd-7be1-488c-81e3-2a4a7b6c779d,0x800,0x64000)/File(\EFI\fedora\shim.efi)\.f.w.u.p.x.6.4...e.f.i...
> Boot0002* UEFI: PM951 NVMe SAMSUNG 512GB, Partition 1	HD(1,GPT,597ea7fd-7be1-488c-81e3-2a4a7b6c779d,0x800,0x64000)/File(EFI\boot\bootx64.efi)..BO
```

```

vathpela, 2017-05-19:
It doesn't matter the order, so long as BootNext has the value of the Linux-Firmware-Updater entry before you reboot.

vathpela, 2017-05-19:
>  Both TPM and FW updated at the same time? I wonder if that's possibly related (it's new that the updates for TPM are available).

I think there is an issue with 2 firmwares available, specifically when one isn't in the ESRT, where we're only applying one of them, but BootNext isn't set the next time, so only one of them is getting applied until the next time you try to apply the update from Linux.  I've seen it on my test box, so I think I'll be able to debug it here.

luzpaz, 2017-07-24:
https://github.com/hughsie/fwupd/issues/100#issuecomment-313311775

thomwiggers, 2017-09-06:
fwupdate fails on my Dell XPS 13 (9350) as follows:

```
 ~  sudo fwupdate -v -a d433959e-03ca-524b-92b7-5022eff81a31 f7375df3c5f903f55ffd64e9ce891da3aa535355-DellTpm1.2_Fw5.81.2.1.cab
firmware resource not found
error trace:
 efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-33773727-8ee7-4d81-9fa0-57e8d889e1fa-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
 lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
```

With force:
```
 ~  upd  sudo fwupdate -v -f -a d433959e-03ca-524b-92b7-5022eff81a31 f7375df3c5f903f55ffd64e9ce891da3aa535355-DellTpm1.2_Fw5.81.2.1.cab
Could not set up firmware update: No such file or directory
error trace:
 efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-d433959e-03ca-524b-92b7-5022eff81a31-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
 lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
 libfwup.c:1207 get_fd_and_media_path(): mkostemps(/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-KhQgeZ.cap) failed: No such file or directory
```

The latter path is also pretty weird to me.

thomwiggers, 2017-10-18:
The `/boot/efi/EFI/boot/efi/EFI` issue is caused by a patch applied by Arch Linux: https://bugs.archlinux.org/index.php?do=details&action=details.addvote&task_id=55815

superm1, 2017-10-18:
Yeah, this should be closed, it's an arch issue caused by arch packaging.

superm1, 2017-10-18:
thanks @thomwiggers

fredoche, 2017-10-18:
I'm still having a similar issue with fedora though (certainly not the same bug but same symptom) .
`/boot/efi/EFI/EFI/fedora/fw/fwupdate-Umq4LN.cap file not found`
I've posted here as well https://github.com/rhboot/fwupdate/issues/80

ArchangeGabriel, 2017-10-18:
@superm1 Note that the ArchLinux issue is different from what is reported here. ;)

fredoche, 2017-11-09:
same kind of error, but different today, on an updated fedora 27:
/builddir/build/BUILD/fwupdate-9/linux/libfwup.c:1224 get_fd_and_media_path(): mkostemps(/boot/efi/EFI//boot/efi/EFI/fedora/fw/fw/fwupdate-4gVoYT.cap) failed: No such file or directory

superm1, 2018-01-08:
This should be closed, everything is fixed in fwupdate 10.

closed 2018-01-09 by superm1

fredoche, 2018-01-09:
Sorry to re-open, however I still cannot update to firmware 1.5.1.
```
[root@cinderblock fred]# fwupdmgr update
Downloading 0.1.5.1 for XPS 13 9350 System Firmware...
Updating 0.1.5.1 on XPS 13 9350 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
```

so it looks fine, but on reboot, I see a very quick test related to the update (I can't really read it, it's too fast), and then the system reboot without performing the bios update.
The subsequent reboot loads fedora normally.

superm1, 2018-01-09:
@fredoche Have you updated to the newer fwupdate version referenced in that RHBZ bug?

fredoche, 2018-01-09:
I have fedora's fwupdate-10-0.2.fc27.x86_64 package installed. It looks like its the version you're mentionning, though I may be mistaken?

fredoche, 2018-01-09:
unfortunately I don't know how to output version directly from fwupdate invocation - I tried "-v" and "--version"

superm1, 2018-01-09:
You need fwupdate-10-1.fc27

luzpaz, 2018-01-09:
Off-topic, IMO it is kind of weird that fwupdate doesn't have a -v  or --version option

superm1, 2018-01-09:
@luzpaz Great point.  I've filed a bug to track this.
https://github.com/rhboot/fwupdate/issues/92
Patches Welcome (™)

fredoche, 2018-01-12:
I tried again using the latest package in fedora's updates-testing repo:
Source       : fwupdate-10-1.fc27.src.rpm
Dépôt        : @System

Again, the computer reboots, then shows two lines of text , one starting with "Searched..", 
and reboots after 5-6 seconds of dark screen.
The two lines of text flash briefly, I don't have time to read... I whish there were a couple seconds of wait time before switching from logging to whichever next step, so I may read and make a better report. I'll try again to read the text and eventually will report back there.

superm1, 2018-01-12:
@fredoche There is suspected to be a second problem with two updates running simultaneously.  Are you by chance running both TPM and UEFI update at the same time?

fredoche, 2018-01-12:
Allright, so, I tried three times yesterday, it went like I described in the previous message, and just after, went for a last try and it JUST worked ... So, yeah, its a bit unsettling, but ok in the end :)
The process updates intel ME, USB controller and maybe a couple of other things.
I don't believe it updated the TPM at the same time this time, though I may have missed something.
And btw gnome-software never showed an available firmware update, I went directly to fwupdmgr.

fredoche, 2018-01-12:
I can run a couple of commands to eventually help you though

superm1, 2018-01-12:
Great!

There is a separate problem with gnome-software not showing updates when used with fwupd 1.0.3 that is being fixed right now.  it will be fixed in the next fwupd 1.0.4 release.
See https://bugzilla.gnome.org/show_bug.cgi?id=792136 for more information.

So I'm going to keep this one closed for now.

If anyone else comes across this same issue as those in here, please do the following:
1) Please make sure you are on latest updates for fwupd/fwupdate from you distro.
2) Follow debug information here: https://github.com/rhboot/fwupdate/wiki/Debugging-UEFI-Capsule-updates
3) File a new issue if you're having problems after following that information.