rhboot/fwupdate #66: fwupdate should provide a display capsule

vathpela, 2017-05-19:
fwupdate should provide a rasterized text image capsule, so that the firmware's message can be localized the same as the OS.

We need a couple of things for this:
1) how do we pick which locale to rasterize into
2) what should it say?
3) what are the dimensions it needs to fit in, what resolution, etc.
4) how do we rasterize it :)

hughsie, 2017-05-20:
Could fwupd pass a string? We already have translation support and the various translator teams set up. I'd even be okay with me passing a BMP data blob if you give me some specs.

vathpela, 2017-06-01:
If you can get the string translated, I can build the rasterized images during the fwupdate build fairly easily, I think.  So that will work fine.

vathpela, 2017-06-01:
"Docs" are here:
- https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/user-experience-for-uefi-firmware-updates
- https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/boot-screen-components

superm1, 2018-02-19:
So once https://github.com/rhboot/fwupdate/pull/98 is reviewed/merged, I expect that at least for Dell platforms this UX capsule handling should be fixed.

Here's my proposal for an action plan:
0. review/merge that PR
1. flip the default in master (eg make a UX capsule by default)
2. Add a field to the LVFS report for fwupd that captures whether UX capsule is supported.
3. Do a call for testing with that version of fwupd and a git snapshot of fwupdate.  Specifically ask if there are problems to report here.  Success reports we should be able to count at LVFS with a custom filter (maybe known issues may work - just mark it to look for Success reports with that criteria)
4. Review LVFS reports to find any reported problems

commit 9493b9b5 (2018-06-11): libfwup: Invert the defaults for creating a UX capsule (Fixes #66)
closed 2018-06-11 by vathpela