rhboot/fwupdate #69: fwupdate

enunes, 2017-06-19:
Trying to use fwupdate on a machine running RHEL, I got:

```
# fwupdate -a 34578c72-11dc-4378-bc7f-b643866f598c firmware.cap 
Could not set up firmware update: No such file or directory
```

Using strace to find out which path caused the error:

```
open("/boot/efi/EFI/fedora/fw/fwupdate-yML5Cm.cap", O_RDWR|O_CREAT|O_TRUNC|O_CLOEXEC, 0600) = -1 ENOENT (No such file or directory)
write(2, "Could not set up firmware update", 32Could not set up firmware update) = 32
```

The "/boot/efi/EFI/fedora" path doesn't make much sense since the machine is running RHEL.

There is this fwupdate variable which may be causing the bug since it contains a "fedora" path, likely from an earlier attempt to use fwupdate on this machine.

```
# hexdump -C /sys/firmware/efi/efivars/fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416
00000000  07 00 00 00 07 00 00 00  72 8c 57 34 dc 11 78 43  |........r.W4..xC|
00000010  bc 7f b6 43 86 6f 59 8c  00 00 02 00 00 00 00 00  |...C.oY.........|
00000020  00 00 00 00 e1 07 04 0c  11 01 1e 00 00 00 00 00  |................|
00000030  ff 07 00 00 02 00 00 00  04 01 2a 00 01 00 00 00  |..........*.....|
00000040  00 08 00 00 00 00 00 00  00 a0 0f 00 00 00 00 00  |................|
00000050  92 a4 82 63 f4 d9 af 42  a7 9d 9a ed 2b 49 4b 8b  |...c...B....+IK.|
00000060  02 02 04 04 4a 00 5c 00  45 00 46 00 49 00 5c 00  |....J.\.E.F.I.\.|
00000070  66 00 65 00 64 00 6f 00  72 00 61 00 5c 00 66 00  |f.e.d.o.r.a.\.f.|
00000080  77 00 5c 00 66 00 77 00  75 00 70 00 64 00 61 00  |w.\.f.w.u.p.d.a.|
00000090  74 00 65 00 2d 00 79 00  4d 00 4c 00 35 00 43 00  |t.e.-.y.M.L.5.C.|
000000a0  6d 00 2e 00 63 00 61 00  70 00 00 00 7f ff 04 00  |m...c.a.p.......|
000000b0
```

Removing the variable fixed the error and I was able to run fwupdate.

superm1, 2017-06-19:
This was supposed to have been cleaned up by the cleanup install script when fwupdate was installed:
https://github.com/rhboot/fwupdate/blob/master/linux/cleanup.in#L22

vathpela, 2017-07-17:
So, I think what this means is the oneshot service never got enabled?

In any case, some work needs to get done on handling extra variables better, both in the API and on the fwupdate.efi end, in order to reasonably handle FMP and also "fwupdate -f -a ...".  So this is probably worth leaving open, to track if that work will obviate the need for the cleanup service.

superm1, 2017-07-17:
@vathpela At least on the RHEL instance @enunes was talking about when I looked at it it seemed to me the oneshot service was being executed but failed to run do to the various EFI variables not being marked immutable (the chattr attempt didn't work).

tchernomax, 2017-09-04:
pull request #77 allow user to bypass this variable.

superm1, 2018-01-09:
I believe that between the PR #77 and 022f2b6 we can call this one closed now.

closed 2018-01-09 by superm1