rhboot/fwupdate #71: fwupdate called in error, work laptop unbootable

vadi2, 2017-07-01:
I've originally reported this over at https://github.com/hughsie/fwupd/issues/152 and with continued research think this is a better place for it. 

I've applied two firmware updates using gnome software centre, but now the laptop is unbootable. How can I fix this?

vadi2, 2017-07-01:
[image]

jmfgdev, 2017-07-01:
What stage of boot is that?

Bios or OS?

vadi2, 2017-07-01:
I'm not sure - this is after the Dell password needs to be put in and
before the Ubuntu encryption password.

jmfgdev, 2017-07-01:
Could you try a live usb?

vadi2, 2017-07-01:
Try what?

jmfgdev, 2017-07-01:
A USB with a live system on it, like for example Ubuntu

vadi2, 2017-07-01:
What would I do with it? Sorry, you haven't specified.

jmfgdev, 2017-07-01:
That would let you know how far the system can get in booting under the circumstances

vadi2, 2017-07-03:
Resolved in https://github.com/hughsie/fwupd/issues/152 - the boot loader was incorrect. Thanks for looking at this!

closed 2017-07-03 by vadi2