rhboot/fwupdate #74: add_capsule forces PERSIST_ACROSS_RESET and INITIATE_RESET flags

semihalf-szczepanek-bartosz, 2017-07-18:
Hi,

according to UEFI specification, CAPSULE_FLAGS_PERSIST_ACROSS_RESET and CAPSULE_FLAGS_INITIATE_RESET flags are optional ([spec](http://www.uefi.org/sites/default/files/resources/UEFI%20Spec%202_6.pdf) page 1097), but fwupdate forces these before passing capsules to UpdateCapsule ([1](https://github.com/rhboot/fwupdate/blob/master/efi/fwupdate.c#L714), [2](https://github.com/rhboot/fwupdate/blob/master/efi/fwupdate.c#L733)). What is the reasoning behind that? Commit message doesn't help.

Thanks,
Bartosz

vathpela, 2017-07-19:
So, the problem here is that there's no good way to discover what fwupdate should set.  Additionally, on x86 machines we've had the goal to be 100% compatible with microsoft's behavior, because we know that's all that will ever really be tested, and in https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/processing-updates , they say:

> The Windows OS loader creates capsule headers using the information from each update’s corresponding ESRT entry, which describes the GUID and flags to use when calling UEFI UpdateCapsule. In setting each capsule header’s flags field, the Windows OS loader always sets CAPSULE_FLAGS_PERSIST_ACROSS_RESET and CAPSULE_FLAGS_INITIATE_RESET. The Windows OS loader may additionally set CAPSULE_FLAGS_POPULATE_SYSTEM_TABLE for firmware types DEVICE_FIRMWARE, if the capsule flag was specified in the INF for the driver package. Proprietary capsule flags may also be specified in the INF and when specified will additionally be included when calling UEFI UpdateCapsule

So we also always set those two flags (though I haven't properly plumbed through the ability to get flags from the .inf as well, yet.)

superm1, 2017-07-19:
Plumbing through the stuff to get that information from the INF is no joke either.  It would mean changing things across 3 projects.
* appstream-glib needs to parse it, 
* fwupd needs to store it in the database, and 
* pass it to libfwup/fwupdate somehow when flashing it.

hughsie, 2017-07-19:
So what exact data needs to be propagated? I'd much rather have metadata NeedsSystemTable=true in the XML file (and some guidance on when to set it...), but this obviously needs to be extendable in the future.

superm1, 2018-01-09:
Since Windows is always setting those flags and this project is aiming for compatibility with Windows capsules I'm closing this issue.  If later we discover a reason that they shouldn't always be set we can revisit this.

closed 2018-01-09 by superm1