rhboot/fwupdate #79: ux-capsule doesn't seem to work on Dell hardware

vathpela, 2017-09-12:
I have code here: https://github.com/rhboot/fwupdate/tree/ux-capsule
It looks (to me) like it should work, but when I apply an update that includes the UX capsule in the list, the machine just reboots without applying any updates.

@superm1 , can I get a bit of help here?

vathpela, 2017-09-12:
You'll need fwupd-labels and basically just do:
`zcat /usr/share/locale/en/LC_IMAGES/fwupd-1920-1080.bmp.gz > fwupd-1920-108.bmp
fwupdate -a 3b8c8162-188c-46a4-aec9-be43f1d65697 fwupd-1920-1080.bmp
fwupdate -a 33773727-8ee7-4d81-9fa0-57e8d889e1fa XPS_9350_1.4.17-firmware.bin`

For some reason BGRT doesn't show up sometimes during boot, in that case you'll need -f on the first fwupdate invocation.

superm1, 2017-09-12:
I'm not super familiar with how this flow is supposed to work.  Can you share your understanding so I can dig on my side?

Does it go something like this:
1. Reboot (Linux-Firmware-Updater is BootNext)
2. shim loads `fwupx64.efi`
3. `fwupx64.efi` loads the pretty bmp from a capsule telling you it's about to do an update
4. `fwupx64.efi` loads the actual content and calls UpdateCapsule
5. BIOS takes over, Dell logo and regular flash process ensue.

Or is it something different than that?

vathpela, 2017-09-20:
It's actually fwupx64.efi loading both capsules into the linked list and doing UpdateCapsule(), and then the firmware displaying the bmp in the text under the BGRT area.  The idea is that the OS has localized text that the firmware doesn't.

I'm actually perfectly okay with you /ignoring/ these capsules, but what seems to happen right now is that if they're present, all capsules are ignored.  Or possibly I'm still malforming something and that's causing them to be ignored.

superm1, 2017-09-20:
From the code on our side that I checked, we just ignore the capsule that matches the GUID corresponding to the UX capsule.  I'm _guessing_ something is still malformed on your side still.

superm1, 2017-09-20:
Any chance you can check on something with a more verbose debug BIOS publicly available (such as an SDP/CRB board?)

superm1, 2018-01-08:
I was hoping that https://github.com/rhboot/fwupdate/commit/7f0b7613c94c7a25dfffa7a1d7f1dcbd07cefc0e fixed this, but I tested and still it's a no go.

I've tried to step through all of your code and found one instance the header looks wrong (https://github.com/rhboot/fwupdate/blob/master/linux/libfwup.c#L1649) but even fixing it didn't resolve it.  I'll probably need to get a debug BIOS and look more closely at what the BIOS is doing.

Have you confirmed this works properly on _anything_ else?

commit 3480d523 (2018-02-23): Fix capsule_image_size calculation in header (Fixes: #79)

It should be the combination of:
 the capsule header
 the UX header
 the size of the image file
not some arbitrary 26 bytes header of the bmp.
closed 2018-02-23 by vathpela