rhboot/fwupdate #80: Nothing happens when running fwupdate or fwupdmgr on Dell XPS 15 9560

Bulmanator, 2017-09-14:
I recently purchased a Dell XPS 15 9560 and the BIOS version is 0.1.3.4 instead of the current 0.1.4.0 version. I have been trying to run fwupdate/ fwupdmgr to update the BIOS but running it does not seem to do anything
```
OS: Arch Linux
Kernel: 4.12.12-1-ARCH
fwupdmgr: 0.9.7
efivars: 31-1
efibootmgr: version 15
```
```
$ sudo efibootmgr
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001
Boot0000  Windows Boot Manager
Boot0001* Grub
Boot0002* grub
Boot0008  WindowsIns
```
This does not change upon calling fwupdate or fwupdmgr (no entry for Linux-Firmware-Updater is created)
When running the update commands it will download the correct package and does not seem to exit with any error however nothing happens, upon calling the commands again after it will say that the device already has an update scheduled

Output:
```
$ sudo fwupdmgr update -v
Downloading 0.1.4.0 for XPS 15 9560 System Firmware...
(fwupdmgr:10145): Fu-DEBUG: creating path /root/.cache/fwupdmgr
(fwupdmgr:10145): Fu-DEBUG: skpping download as file already exists
Updating 0.1.4.0 on XPS 15 9560 System Firmware...
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         [-                                       ]
Decompressing…         [ -                                      ]
Decompressing…         [  -                                     ]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [idle]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [scheduling]
Scheduling…            [   -                                    ]
Retrying as an offline update...
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [decompressing]
Decompressing…         [   -                                    ]
Decompressing…         [    \                                   ]
Decompressing…         [     \                                  ]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [idle]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [scheduling]
Scheduling…            [      \                                 ]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::status-changed() [idle]
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::device-changed((null))
(fwupdmgr:10145): Fwupd-DEBUG: Emitting ::changed()
```
Note: This did not print correctly on my terminal, I had to pipe it to a file and delete H^ strings in vim  

Output fwupdmgr get-devices:
```
$ sudo fwupdmgr get-devices
Intel AMT (unprovisioned)
  DeviceID:             /dev/mei
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Plugin:               amt
  Flags:                internal|registered
  Version:              11.6.29
  VersionBootloader:    11.6.29
  Created:              2017-09-14

XPS 15 9560 TPM 2.0
  DeviceID:             DELL-bc891321-25aa-54f6-86e7-633d065b8d5dlu
  Guid:                 bc891321-25aa-54f6-86e7-633d065b8d5d
  Plugin:               dell
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              1.3.1.0
  Created:              2017-09-14

XPS 15 9560 TPM 1.2
  DeviceID:             DELL-c5e6d7ef-4270-54bf-af0f-8b8ef2a561e6lu
  Guid:                 c5e6d7ef-4270-54bf-af0f-8b8ef2a561e6
  Plugin:               dell
  Flags:                internal|require-ac|locked|registered
  Created:              2017-09-14

XPS 15 9560 System Firmware
  DeviceID:             UEFI-34578c72-11dc-4378-bc7f-b643866f598c-dev0
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Description:          <p>Updating the system firmware improves performance.</p>
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.3.4
  VersionLowest:        0.1.3.4
  Created:              2017-09-14

Integrated Webcam HD
  DeviceID:             usb:00:0c
  Guid:                 4c03e6af-b94c-5c18-8689-e77ceadbe524
  Guid:                 fb0df457-b15e-5d4a-b73a-8491dca96a07
  Plugin:               usb
  Flags:                registered
  DeviceVendorId:       USB:0x0C45
  Version:              86.5
  Created:              2017-09-14

GP107M [GeForce GTX 1050 Mobile]
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_01_0_0000_01_00_0
  Guid:                 37921484-dff0-57a0-b7dd-4cd6a82b54a1
  Plugin:               udev
  Flags:                internal|registered
  DeviceVendor:         NVIDIA Corporation
  DeviceVendorId:       PCI:0x10DE
  Created:              2017-09-14

Unknown Device
  DeviceID:             ro__sys_devices_pci0000_00_0000_00_02_0
  Guid:                 3ab5fe9e-bef7-5421-920a-ef7301884933
  Plugin:               udev
  Flags:                internal|registered
  DeviceVendor:         Intel Corporation
  DeviceVendorId:       PCI:0x8086
  Created:              2017-09-14
```
The XPS 15 9560 System Firmware with GUID: 34578c72-11dc-4378-bc7f-b643866f598c is the device which requires updating

And applying it manually through fwupdate outputs this:
```
$ sudo fwupdate -a 34578c72-11dc-4378-bc7f-b643866f598c ~/.cache/fwupdmgr/5b8e9cbc79dd4f37df0f792196f246d9b7da1e0d-firmware_XPS_9560_1_4_0.wu.cab -v
Could not set up firmware update: No such file or directory
error trace:
efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
libfwup.c:1194 get_fd_and_media_path(): failed to make /boot/efi/EFI/arch/fw: No such file or directory
```

If I have missed any important information please let me know and I will provide the details

arvids, 2017-09-22:
Hi,
Seems I have the same machine and the same issue, did you find a solution?

The only difference seems to be that applying the update through fwupdate manually my message is slightly different:

`% sudo fwupdate -a 34578c72-11dc-4378-bc7f-b643866f598c .cache/fwupdmgr/e69073e5403ee433ddfdd40b53781e91074b07b2-firmware_XPS_9560_1_5_0.wu.cab -v
Could not set up firmware update: No such file or directory
error trace:
 efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
 lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
 efivarfs.c:230 efivarfs_get_variable(): open(/sys/firmware/efi/efivars/fwupdate-34578c72-11dc-4378-bc7f-b643866f598c-0-0abba7dc-e516-4167-bbf5-4d9d1c739416): No such file or directory
 lib.c:152 efi_get_variable(): ops->get_variable failed: No such file or directory
 libfwup.c:1207 get_fd_and_media_path(): mkostemps(/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-9FJxpS.cap) failed: No such file or directory
`

superm1, 2017-09-22:
@Bulmanator :I'm pretty sure this is the same ole' arch package doesn't make directories problem that keeps getting reported over and over.
https://github.com/hughsie/fwupd/wiki/Common-Problems

@arvids Your problem is that the arch package is not configured to the correct ESP path.  Look closer at the error and you'll see it's got the same thing twice. 
`/boot/efi/EFI//boot/efi/EFI/arch/fw/fw/fwupdate-9FJxpS.cap`
Report it to @archangelgabriel.

arvids, 2017-09-28:
Sorry for the late reply @superm1 , where should I report it to @ArchangelGabriel?

superm1, 2017-09-28:
@arvids https://bugs.archlinux.org/?project=5&cat%5B%5D=33&string=fwupdate

arvids, 2017-09-30:
Thank you, I  have reported the bug.

fredoche, 2017-10-14:
same here probably:
11:10:17:0089 Gs  failed to call gs_plugin_update_app on fwupd: UEFI firmware update failed:
{error #0} /builddir/build/BUILD/fwupdate-9/linux/libfwup.c:1193 get_fd_and_media_path(): open of /boot/efi/EFI/EFI/fedora/fw/fwupdate-Umq4LN.cap failed: No such file or directory

two times EFI in the path

superm1, 2017-10-18:
@fredoche you're on Fedora and that's happening?  Looks like it's reported against F27 here too: https://bugzilla.redhat.com/show_bug.cgi?id=1495137

I think @vathpela needs to double check the paths to make sure the Arch bug didn't creep into fedora as a result too.

HaasJona, 2017-10-26:
Same duplicate EFI in the path on Fedora 27 and on a Dell XPS 13 9350

mbellgb, 2017-10-27:
I'm using Arch, appears to be using `/boot/efi/efi/efi/EFI/arch/fw/fwupdate-XXXXXX.cap`. Possibly due to a different issue since my ESP is mounted at `/boot` rather than `/boot/efi`?

ArchangeGabriel, 2017-10-27:
@mbellgb Having your EFI at a different mount point than `/boot/efi` is indeed an issue that will be fixed in `fwupdate` next release (`10`) by allowing you to specify the path to use.

For people on Arch using `/boot/efi` as mount point, please try with `fwupdate 9-3` from `[community-testing]` and report.

ledbettj, 2017-11-03:
I've installed 9-3 from community-testing; I have my boot partition mounted at `/boot`:
```
/dev/nvme0n1p1 on /boot type vfat 
```

```
/boot/
├── EFI
│   ├── arch
│   │   ├── fw
│   │   └── fwupx64.efi
│   ├── BOOT
│   │   └── BOOTX64.EFI
│   ├── Dell
│   │   ├── Bios
│   │   │   └── Recovery
│   │   │       ├── BIOS_CUR.RCV
│   │   │       └── BIOS_PRE.rcv
│   │   └── logs
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
│   ├── EFI
│   │   └── arch
│   │       ├── fw
│   │       └── fwupx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   └── arch.conf
│   └── loader.conf
└── vmlinuz-linux

14 directories, 14 files
```

Note that I have fwupdate installed in both `/boot/EFI` and `/boot/EFI/EFI`.

Trying to install latest fw gives me:

```
No upgrades for device, current is 1.3.1.0: 1.3.1.0=same
Downloading 0.2.3.1 for XPS 13 9360 System Firmware...
Updating 0.2.3.1 on XPS 13 9360 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [    \                                  ]UEFI firmware update failed: {error #0} libfwup.c:1178 get_fd_and_media_path(): open of /boot/efi/efi/efi/efi/EFI/arch/fw/fwupdate-keAknA.cap failed: No such file or directory
```

Is there something else I need to try to git rid of the `/efi/` echo? 😁

superm1, 2017-11-03:
@ArchangeGabriel  ^

fredoche, 2017-11-09:
@superm1 I tried again today on fedora 27 and I get this:
https://github.com/rhboot/fwupdate/issues/62#issuecomment-343098978

superm1, 2017-11-09:
@fredoche Fedora has a bug that needs to be fixed.
https://bugzilla.redhat.com/show_bug.cgi?id=1506609 (duplicate: https://bugzilla.redhat.com/show_bug.cgi?id=1495137)

ArchangeGabriel, 2017-11-26:
@ledbettj Sorry for the delay. You’re affected by the fact `/boot/efi` is assumed as ESP mount point for now. You’ll have to wait for the next fwupdate release (10).

ledbettj, 2017-11-26:
> @ledbettj Sorry for the delay. You’re affected by the fact /boot/efi is assumed as ESP mount point for now. You’ll have to wait for the next fwupdate release (10).

Ahh, sorry, I misread the previous comment as saying that 9-3 would resolve this issue.  My fault!

vathpela, 2018-01-08:
I think this should be fixed in the current master; please confirm if you can.

superm1, 2018-01-09:
@ArchangeGabriel I believe this will just require you to package up version 10 now.  From an upstream perspective this is fixed (I personally verified on Ubuntu and others have on Fedora).  I'm closing up this issue.  If someone encounters problems on Arch that lead you to this bug, please test with version 10 and open a new bug.  You can feel free to reference this bug in your description though.

closed 2018-01-09 by superm1

superm1, 2018-01-12:
If anyone else comes across this same issue as those in here, please do the following:

Please make sure you are on latest updates for fwupd/fwupdate from you distro.
Follow debug information here: https://github.com/rhboot/fwupdate/wiki/Debugging-UEFI-Capsule-updates
File a new issue if you're having problems after following that information.