rhboot/fwupdate #86: Fail to reboot into Ubuntu after successful update

galgalesh, 2017-10-22:
I triggered the update with Gnome Software on Ubuntu 17.10. Gnome Software asked me to restart, I click restart. During boot, the update starts correctly, and succeeds. After it succeeds, the laptop boots back into fwupd, which fails with the following error:

```
found update fwupdate-49w03513-7afe-49a-9d03-89e5ba1da865-0
fwupdate: No updates to process. Called in error?
start_image() returned Invalid Parameter
```

After a few seconds, it shows another screen:

```
No bootable devices found.
Press F1 key to retry boot.
Press F2 key to run onboard diagnostics.
Press F5 key to run onboard diagnostics.
```

I was able to workaround the issue by going into `setup`, `general > Boot Sequence`, and adding `grub64.efi` back as primary boot option.

superm1, 2017-10-22:
Can you please share more details about your machine?  What model number? BIOS version you were at and version you were trying to upgrade to?

galgalesh, 2017-10-22:
Dell Latitude 5580, I was at 1.4.3 and upgraded to 1.6.4.

I've been trying to replicate this by downgrading and upgrading a few times, but I'm not able to replicate it anymore.

superm1, 2017-10-23:
Hmm interesting.  to your first comment, a little more clarification:
When you say it starts successfully and succeeds, do you mean it flashed up to 1.6.4 and then didn't want to boot again?  or do you mean that fwupdate opened and that error came up?

galgalesh, 2017-10-23:
It flashed to 1.6.4 and it didn't boot again. (well it booted into those two error messages)

superm1, 2017-10-23:
It sounds to me like you were missing a boot entry initially.  Were you booting using the removable media path \efi\boot\bootx64.efi?

galgalesh, 2017-10-23:
I don't know. It was a standard Ubuntu 17.10 installation.

Note that this was not the first upgrade I performed. The first one succeeded without issues, so I don't think it's an installation issue.

I downgraded the firmware manually using a USB stick and Dell's builtin firmware update utility, in order to reproduce some issues with the fwupdate integration in the software center.

superm1, 2017-10-23:
Since this was with ubuntu 17.10, there's actually evidence logged into the journal log for fwupd.  can you look through journalctl to see if you have the `efibootmgr `output from before the update was performed?  It's been in fwupd 0.9.6 and later (17.10 has 0.9.7).

(Came from this commit: ) https://github.com/hughsie/fwupd/commit/f043f1f94b7f671e7e8d4fa3963a47e045d94f59

galgalesh, 2017-10-24:
No such luck, apparently Ubuntu 17.10 still doesn't enable persistent logs. Only the logs since last boot are visible. I configured persistent logs now, but that won't help much for this bug..

```
sudo journalctl --list-boots
 0 7318355f8d464ff7ae7d8f967466edd0 Tue 2017-10-24 09:11:37 CEST—Tue 2017-10-24 09:26:31 CEST
```

superm1, 2017-10-24:
Ah well darn.

gerobum, 2017-11-06:
Is this topic closes ? Because I have the same issue. Yesterday, I answered "yes" to the "update manager" which ask to me if I would like to change my firmware. And this morning, unable to boot on my ubuntu 
found update fwupdate-49w03513-7afe-49a-9d03-89e5ba1da865-0
fwupdate: No updates to process. Called in error?
start_image() returned Invalid Parameter

galgalesh, 2017-11-06:
@gerobum Do you have persistent logs on?

For reference,

- The fwupd bug in the Ubuntu launchpad: https://bugs.launchpad.net/fwupd/+bug/1730343
- The bug that persistent logs are disabled by default: https://bugs.launchpad.net/ubuntu/+source/systemd/+bug/1618188

gerobum, 2017-11-06:
@galgalesh I don't what do you by "persistent logs on". But I don't know what to do because I am unable to boot.

Have you fixed your problem ?

galgalesh, 2017-11-06:
Oh, why didn't you say that earlier! I added a picture guide to my original message on the top of this thread.

Can you show the output of

```bash
 sudo journalctl --list-boots
```

to figure out if you have persistent logs enabled?

gerobum, 2017-11-06:
@galgalesh Ok thanks. But I tried to do the same. Unfortunatly I have now directly this
No bootable devices found.
Press F1 key to retry boot.
Press F2 key to run onboard diagnostics.
Press F5 key to run onboard diagnostics.

galgalesh, 2017-11-06:
@gerobum can you show a picture of the "boot sequence screen"? (the last picture in my tutorial)

galgalesh, 2017-11-06:
@gerobum a picture of the "boot menu" is also useful. You get that screen by furiously tapping <kbd>esc</kbd> after rebooting the laptop. Example shown below

[image]

gerobum, 2017-11-06:
Thanks a lot. It works now.
When I type  sudo journalctl --list-boots, I have something like that
0 f4998...ae3b2f6 Mon 2017-11-06 14:57:59 CET—Mon 2017-11-06 1
But I didn't know what doest it mean.

Anyway. Thanks.

galgalesh, 2017-11-06:
That means the systemd journal (the log collector) only contains logs since the last boot (which happened on `Mon 2017-11-06 14:57:59`).

Thus, we cannot look at the logs from when `fwupd` failed, because those logs were not saved.

gerobum, 2017-11-06:
I see.

seb128, 2017-11-06:
the journal isn't persistant but the logs are stored in /var/log/syslog as well no? that should have records from previous boot (it's being rotated so could be .1)

jeremybmerrill, 2017-11-17:
Chiming in to say that I had this same bug occur on my Dell XPS 13 (2016), but didn't have persistent logs on. However @galgalesh, your guide at the top of the thread worked perfectly to fix my issue, so thank you!

superm1, 2017-11-17:
@jeremybmerrill were you on Ubuntu as well?  We need someone to help to verify the SRU that has been filed.

jeremybmerrill, 2017-11-17:
@superm1 Yes, Ubuntu 17.10. Happy to do what I can to help, just tell me what to do.

galgalesh, 2017-11-17:
@superm1 I think something went wrong, the patch is still not in proposed..

```
sudo apt-get install fwupdate-signed/artful-proposed
[sudo] password for merlijn: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Release ‘artful-proposed’ for ‘fwupdate-signed’ was not found
```

superm1, 2017-11-17:
@jeremybmerrill Yes so this is the public bug: https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1730343
Can you follow the directions on installing the fwupdate-signed and fwupdate packages from proposed?

@galgalesh 
Hmm, I see them in proposed according to: https://launchpad.net/ubuntu/+source/fwupdate and https://launchpad.net/ubuntu/+source/fwupdate-signed

Presumably you've already apt updated and have proposed activated?

jeremybmerrill, 2017-11-18:
@superm1 Do you know if the bug is such that my system did update the firmware properly, but just didn't boot up correctly afterwards? Or did the system not update the firmware?

I enabled proposed, ran `sudo apt-get install fwupdate/artful-proposed` (and tried to run `sudo apt-get install fwupdate-signed/artful-proposed`, but it said `E: Release 'artful-proposed' for 'fwupdate-signed' was not found` -- same as @galgalesh ).

How do I run the update again, now that I have a new `fwupdate`?

superm1, 2017-11-19:
@jeremybmerrill yes that's the bug, it didn't boot up afterwards properly.

To confirm you have the latest package can you please run:
`apt-cache policy fwupdate-signed`

IIRC artful-proposed isn't enabled by default, you have to modify `/etc/apt/sources.list` to do it.

After you confirm you have the new package, you can manually download the CAB file from LVFS for your system.

then run:

`fwupdmgr install --allow-reinstall blah.cab`

galgalesh, 2017-11-19:
```
$ apt-cache policy fwupdate-signed
fwupdate-signed:
  Installed: (none)
  Candidate: 1.14+9-2
  Version table:
     1.14+9-2 500
        500 http://us.archive.ubuntu.com/ubuntu artful/main amd64 Packages
        100 /var/lib/dpkg/status
$ apt-cache policy fwupdate
fwupdate:
  Installed: 9-2ubuntu0.17.10.1
  Candidate: 9-2ubuntu0.17.10.1
  Version table:
 *** 9-2ubuntu0.17.10.1 400
        400 http://us.archive.ubuntu.com/ubuntu artful-proposed/main amd64 Packages
        100 /var/lib/dpkg/status
     9-2 500
        500 http://us.archive.ubuntu.com/ubuntu artful/main amd64 Packages
```

1. I uninstalled `fwupdate-signed` and installed `fwupdate` I suspect the "signed" is only relevant for secureboot?
2. I removed boot options in my UEFI boot menu.
3. I rebooted and it boots into Ubuntu
4. I downloaded the firmware from here: https://fwupd.org/lvfs/devicelist (Are you aware that there are two entries for Latitude 5X80?)
5.  I ran `fwupdmgr install --allow-reinstall <filename>.cab` and rebooted my machine.
6. fwupdate starts, updates, and reboots into Ubuntu.

**So If your assessment of the cause of the bug is correct, then the bug is fixed.**

However there is clearly something wrong with fwupdate-signed since it isn't visible for us in the artful-proposed repos as you can see in the output of the commands up top.

superm1, 2017-11-19:
Can you confirm the efibootmgr -v output after that list of steps you followed?

galgalesh, 2017-11-20:
```
$ efibootmgr -v
BootCurrent: 0002
Timeout: 2 seconds
BootOrder: 0002,0001
Boot0000* Windows Boot Manager	HD(1,GPT,4076bdbf-fe0f-4db3-85f6-2cab5d6263a3,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* UEFI: SanDisk X400 M.2 2280 512GB, Partition 1	HD(1,GPT,50def640-b2e4-47e9-9456-a8755c5b6cd4,0x800,0x100000)/File(EFI\boot\bootx64.efi)..BO
Boot0002* ubuntu	HD(1,GPT,50def640-b2e4-47e9-9456-a8755c5b6cd4,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
```

I am surprised that windows boot manager shows up here, I don't run Windows..

4l1fe, 2017-11-20:
@galgalesh bro, you have saved my life by your guide!

superm1, 2017-11-20:
@4l1fe there was a problem with the building of fwupdate-signed for some reason on Launchpad.  I just requested it to be rebuilt, hopefully that will fix the problem with secure boot.

@galgalesh 
Thanks, so that confirms that the issue I was trying to fix did get fixed properly.  Would you mind adding a comment to Launchpad and the tag that indicates you verified the fwupdate package?  fwupdate-signed we'll need to adjust something.

superm1, 2017-11-20:
@galgalesh regarding your comment about two Latitude 5x80 devices, yes it's intended.  There are two different variants of the 5x80's that contain incompatible BIOS binaries.  In that case you need to match the ESRT GUID to the one you have (which it sounds like you were able to successfully or luckily do).

For most people it doesn't affect them because LVFS sorts it out.  It's just when you manually go hunting for the binaries that it could be problematic.

4l1fe, 2017-11-26:
@superm1 Can you say if my bootloading depends on a new firmware ? coz it became very very slow after updating. 
`$ systemd-analyze 
Startup finished in 6.016s (firmware) + 5.885s (loader) + 7.954s (kernel) + 3min 328ms (userspace) = 3min 20.185s`

And how can i downgrade the firmware and return previous settings in `Boot sequence` is showed in the first topic?  Advice pls

4l1fe, 2017-11-28:
Thx for the advice. I have solved this problem. It was not related to `fwupdate`.

jeremybmerrill, 2017-12-04:
@superm1 : I see that the fwupdate-signed version for artful-proposed is now available in the repos. I installed it and tried to install the updates. The installation failed (due to the TPM being owned, or something? Then due to the update supposedly not being for my system -- but these are not fwupdate related, I don't think), but then it rebooted into Ubuntu properly. I'm not sure if this avoided the spot where the problem was or if it reflects a successful fix.

superm1, 2017-12-04:
@jeremybmerrill can you please confirm your `efibootmgr -v` output?  That will confirm whether the reported problem is fixed or not.

Your other two issues are quite interesting to me though.  I'd like some other details if you can share them.
1. Model number for system
2. `fwupdmgr devices` output
3. Was this a firmware or TPM update you were doing?

jeremybmerrill, 2017-12-06:
```# efibootmgr -v
BootNext: 0001
BootCurrent: 0003
Timeout: 0 seconds
BootOrder: 0003,0000,0001
Boot0000* Windows Boot Manager	HD(1,GPT,35f0cd73-d921-4ceb-a20a-1c1acf5c6761,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....4...............
Boot0001* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,35f0cd73-d921-4ceb-a20a-1c1acf5c6761,0x800,0xfa000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0003* Ubuntu	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-03-00-02-6E-8C)/HD(1,GPT,35f0cd73-d921-4ceb-a20a-1c1acf5c6761,0x800,0xfa000)/File(\EFI\ubuntu\grubx64.efi)
Boot000A* UEFI: THNSN5512GPU7 NVMe TOSHIBA 512GB, Partition 1	HD(1,GPT,35f0cd73-d921-4ceb-a20a-1c1acf5c6761,0x800,0xfa000)/File(EFI\Microsoft\Boot\bootmgfw.efi)..BO
```

1. Dell XPS 13 9350
2. output --> https://gist.github.com/jeremybmerrill/bb3d7a4a169de6d85b0c0ad00efc8604
3. Both (it's what the Software Center suggests). 
[image]

Thanks again for all your help!

jeremybmerrill, 2017-12-12:
@superm1 : Oddly enough, I restarted again (for other reasons) and it seems to have updated the firmware. Not sure what happened the first time.

superm1, 2017-12-12:
Did it update the tpm too? I think this might be order of events.

jeremybmerrill, 2017-12-12:
@superm1 It appears to have only updated the BIOS, but it did not update the TPM. (At least -- the Updater still suggests that I update the TPM)

superm1, 2017-12-12:
I suppose the tpm update won't take due to it being owned now right? Did you take ownership of it?

jeremybmerrill, 2017-12-12:
Yes, I think that's why it failed. I need to figure out how to take
ownership of it, I just haven't had time

superm1, 2017-12-12:
Well so it's not supposed to even offer in Ubuntu if it's owned. That's the odd part to me.

superm1, 2017-12-12:
Well it's not supposed to allow switching modes if owned. I think upgrades within the mode should work though.

KoenDG, 2017-12-14:
Chiming in, just now had the same issue with a Dell Inspiron 3520, running Ubuntu 16.04 LTS.

Ouput of the commands I found in this thread:

```
efibootmgr -v
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0000
Boot0000  Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0001* Diskette Drive	BBS(Floppy,Diskette Drive,0x0)..BO
Boot0002* Ubuntu	PciRoot(0x0)/Pci(0x17,0x0)/Sata(0,65535,0)/HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(\EFI\ubuntu\grubx64.efi)
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC	BBS(Network,IBA CL Slot 00FE v0109,0x0)..BO
Boot0006* M.2 SATA SSD	BBS(HD,P0: Micron 1100 SATA 256GB,0x0)..BO
Boot0007* UEFI: Micron 1100 SATA 256GB, Partition 1	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO

```
```

sudo apt-cache policy fwupdate-signed 
fwupdate-signed:
  Installed: 1.11.1+0.5-2ubuntu5
  Candidate: 1.11.1+0.5-2ubuntu5
  Version table:
 *** 1.11.1+0.5-2ubuntu5 500
        500 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.11+0.5-2ubuntu4 500
        500 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages

sudo apt-cache policy fwupdate
fwupdate:
  Installed: 0.5-2ubuntu5
  Candidate: 0.5-2ubuntu5
  Version table:
 *** 0.5-2ubuntu5 500
        500 http://archive.ubuntu.com/ubuntu xenial-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     0.5-2ubuntu4 500
        500 http://archive.ubuntu.com/ubuntu xenial/main amd64 Packages

sudo apt-cache policy fwupdate-amd64-signed 
fwupdate-amd64-signed:
  Installed: (none)
  Candidate: 1.11.1+0.5-2ubuntu5
  Version table:
     1.11.1+0.5-2ubuntu5 500
        500 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages
     1.11+0.5-2ubuntu4 500
        500 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages
```

and

```
sudo journalctl --list-boots
 0 8c2d64526f214aa184b470acd3b00806 Thu 2017-12-14 20:10:13 CET—Thu 2017-12-14 20:24:16 CET
```

Hope it helps.

superm1, 2017-12-14:
@KoenDG Can you please add a comment to https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1730343

The Ubuntu archive admin needs to release the uploaded package for 16.04.

KoenDG, 2017-12-15:
@superm1 Comment made.

superm1, 2018-01-02:
@KoenDG would you be able to test the 16.04 package as it's been accepted by that admin?

KoenDG, 2018-01-02:
@superm1 I followed the steps found here: https://wiki.ubuntu.com/Testing/EnableProposed

And selected fwupdate and fwupdate-signed from aptitude(which were not both in the same part of the tree for some reason).

3 packages were selected for update, fwupdate, fwupdate-signed and the library behind it, I already forgot the name.

Install showed no errors. I then performed `sudo shutdown -r now` and this had no error. No actual firmware update either, it just restarted the system. same with `shutdown -h`.

`dpkg -l fwupdate | cat` shows version 0.5-2ubuntu6 installed for amd64

I don't know if there's something else I should test. There isn't actually a firmware update available so I don't know how to test it further.

superm1, 2018-01-02:
@KoenDG thanks for looking.  Yes they're two separate packages due to how the signing works in Ubuntu.

You can download the latest CAB file for your system and install with `fwupdmgr install <blah.cab> --allow-reinstall`
If you can succesfully boot after installing that way then I would deem it a successful test (and please mark it on the Launchpad bug too).

KoenDG, 2018-01-02:
@superm1 I can't get it to work.

I had to look for the cab file in my download cache, it was still there. The Dell website isn't offering it for download anywhere.

I did the command as instructed, got the output saying an upgrade was scheduled.

I tried both `shutdown -r` and going to the Ubuntu Software menu and doing the upgrade&restart from there.

I tried it with and without `--allow-reinstall`.

In all instances, a regular reboot happens and nothing else.

`dmidecode` still says the BIOS is on version 1.7.5, dating 11/12/2017

Not sure why it's not picking up the upgrade.

superm1, 2018-01-02:
The CAB file comes from LVFS not from the Dell site, so that's why you couldn't find it there.  It's good you had it in your cache though.

After you have scheduled it but before rebooting, can you please share the output of:
```
# efibootmgr -v
```

In `dmidecode` output the date should stay the same even if you were successful.  It's the date the BIOS was built not the date that it was flashed.  To query the date of the last flash you would have to look elsewhere.

KoenDG, 2018-01-02:
@superm1 As requested:

```
# efibootmgr -v
BootNext: 0008
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0000,0008
Boot0000  Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
Boot0001* Diskette Drive	BBS(Floppy,Diskette Drive,0x0)..BO
Boot0002* ubuntu	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC	BBS(Network,IBA CL Slot 00FE v0109,0x0)..BO
Boot0006* M.2 SATA SSD	BBS(HD,P0: Micron 1100 SATA 256GB,0x0)..BO
Boot0007* UEFI: Micron 1100 SATA 256GB, Partition 1	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(EFI\boot\bootx64.efi)..BO
Boot0008* Linux-Firmware-Updater \fwupx64.efi	HD(1,GPT,9b1d146c-b8a3-45b6-9076-150bb5292428,0x800,0x12c000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.x.6.4...e.f.i...
```

superm1, 2018-01-03:
Hmm, looks like it did the correct thing to me.  And after that you don't see the fwupx64.efi quick between post and GRUB coming up pop up?  
Do you have secure boot enabled?  Can you confirm the checksums of /boot/efi/efi/ubuntu/fwupx64.efi?

KoenDG, 2018-01-03:
Nothing comes up. Secure boot is disabled

MD5: 1a68d9b11532c19640b4a2eeeee4fd80  /boot/efi/efi/ubuntu/fwupx64.efi
SHA256: 52d202634650ae9196a9bcf79c6a689f15feb82be4f4cc5462a0dc1aca64456e  /boot/efi/efi/ubuntu/fwupx64.efi

Could this be related to the fact that I had to use the fix given in the top post here, when this happened to me? So I selected the ubuntu shimx64.efi and that's still selected?

superm1, 2018-01-03:
That is the correct checksum.  Something that stick out is that you have two different Linux Firmware Updater entries.  I wonder if that's throwing something off.

Can you please try to delete both with efibootmgr and try again?

KoenDG, 2018-01-03:
@superm1 I executed `sudo efibootmgr -D` which deletes duplicates.

Attempt the update now gave me a new message: Cannot install update when not on AC power.

So that's progress. I don't have my AC adapter with me so it'll have to wait until tomorrow.

superm1, 2018-01-03:
@KoenDG That message came from the Dell flash screen or from fwupd?  If it's from the Dell flash screen that's great news and it means that the duplicates were the root cause.

KoenDG, 2018-01-03:
@superm1 Sorry, it's from fwupdmgr. The flash screen just saying "Bootin in insecure mode".

```
... .cache/gnome-software/3.20/firmware/4a89bbb6bd769fdfc315c37db55a74b8e7f25768-firmware.Latitude_5x80_Precision_3520_1.8.1H.wu.cab --allow-reinstall
 * Decompressing firmware
Retrying as an offline update...
 * Idle
 * Decompressing firmware
Cannot install update when not on AC power
```

[16 more replies]

closed 2018-01-09 by superm1