rhboot/fwupdate #88: Petitition: AMI (American Megatrends) EFI BIOS (Opti IV and Opti V?) support

timofonic, 2017-11-02:
Hello.

Can you developers consider support for AMI OPti (3,4,4.5,5) support? My computer is a [MSI GE62 2QD Apache Pro](https://www.msi.com/Laptop/GE62-2QD-Apache-Pro.html) and I would like a better experience regarding firmware updating and others.

Relevant links:

http://hargrave.info/articles/afulnx.html
https://github.com/RomanHargrave/amifldrv
https://github.com/sling00/afulnx-dkms-ethos
https://github.com/mrwnwttk/afulnx
https://github.com/csett86/amifldrv/tree/v5.07.01 (code a bit more ordered and fixed)

Reason: It's a mess and it fails on many people:
https://forum.proxmox.com/threads/server-bios-update-using-afulnx-and-building-amifldrv-while-running-froxlor.27380/
https://puri.sm/posts/librem-13-coreboot-report-january-12-2017/ (even a dev has issues with official AMI tools, that's very intriguing and disgusting to know)

Thanks a lot in advance.

skochinsky, 2017-11-02:
I think you meant Aptio, not "Opti" (I don't think that exists). Unfortunately the driver alone is not enough, you need to know the whole protocol of performing the update (e.g. the SMI values  and their arguments). You can possibly discove it by strace-ing the updater but getting it right is not trivial, especially with the recent signed updates and recovery modes. Proper support for it will require participation of AMI engineers because only they know all the things that need to be done for reliable update and how to handle all possible situations (e.g. failures). For now I would suggest you to use built-in updater in the BIOS setup UI (if your board has one).

superm1, 2017-11-02:
This project is about supporting industry standard UEFI capsule update.  It works by the  vendor providing a UEFI capsule in OS and then passing it to a BIOS based firmware flash method.  This project can't do anything around alternative proprietary tools and routines.

If your vendor doesn't support it, I would recommend::
1. You contact them asking them to release UEFI capsule updates in the future and post them [the LVFS
](https://fwupd.org/) for use with Linux machines.
2. Encourage you to vote with your wallet in the future for [vendors that care about Linux](https://fwupd.org/status).

vathpela, 2018-01-08:
This is really an issue for https://github.com/hughsie/lvfs-website rather than for fwupdate itself.

closed 2018-01-08 by vathpela