rhboot/fwupdate #93: FTBFS with new release

ArchangeGabriel, 2018-01-10:
During `make` on ArchLinux:
```
gcc -march=x86-64 -mtune=generic -O2 -pipe -fstack-protector-strong -fno-plt -DFWUPDATE_HAVE_LIBSMBIOS__ -Wall -Wextra -Werror -Wno-error=cpp -Wno-unused-result -Wno-unused-function -Wsign-compare -Werror=sign-compare -fshort-wchar --std=gnu11 -DLOCALEDIR=\"/usr/share/locale\" -D_GNU_SOURCE -DFWUP_EFI_DIR_NAME=\"arch\" -DFWUP_ESP_MOUNTPOINT=\""/boot/efi"\" -I include -I /build/fwupdate/src/fwupdate-10/linux/include/ -iquote /build/fwupdate/src/fwupdate-10/include/  -I/usr/include/efivar  -I/usr/include/efivar    -fPIC -c -o libfwup.o libfwup.c
xgettext -a --package-name=libfwup --package-version=10 -o libfwup.po libfwup.c
libfwup.c: In function ‘make_ux_capsule_entry’:
libfwup.c:733:30: error: ‘efi_guid_ux_capsule’ undeclared (first use in this function); did you mean ‘efi_guid_ux_capsule_guid’?
  fwup_ux_capsule.esre.guid = efi_guid_ux_capsule;
                              ^~~~~~~~~~~~~~~~~~~
                              efi_guid_ux_capsule_guid
libfwup.c:733:30: note: each undeclared identifier is reported only once for each function it appears in
libfwup.c: In function ‘write_ux_capsule_header’:
libfwup.c:1648:11: error: ‘efi_guid_ux_capsule’ undeclared (first use in this function); did you mean ‘efi_guid_ux_capsule_guid’?
   .guid = efi_guid_ux_capsule,
           ^~~~~~~~~~~~~~~~~~~
           efi_guid_ux_capsule_guid
libfwup.c: In function ‘fwup_set_up_update’:
libfwup.c:1795:37: error: ‘efi_guid_ux_capsule’ undeclared (first use in this function); did you mean ‘efi_guid_ux_capsule_guid’?
  if (!efi_guid_cmp(&re->esre.guid, &efi_guid_ux_capsule)) {
                                     ^~~~~~~~~~~~~~~~~~~
                                     efi_guid_ux_capsule_guid
libfwup.c: In function ‘fwup_set_up_update_with_buf’:
libfwup.c:1899:37: error: ‘efi_guid_ux_capsule’ undeclared (first use in this function); did you mean ‘efi_guid_ux_capsule_guid’?
  if (!efi_guid_cmp(&re->esre.guid, &efi_guid_ux_capsule)) {
                                     ^~~~~~~~~~~~~~~~~~~
                                     efi_guid_ux_capsule_guid
make[1]: *** [/build/fwupdate/src/fwupdate-10/linux/Makefile:90: libfwup.o] Error 1
make[1]: *** Waiting for unfinished jobs....
```

superm1, 2018-01-10:
It needs https://github.com/rhboot/efivar/commit/cd732494ba7685feaf71b9ee58619ca6aef39fc9 on top of efivar 32 at least until a new efivar release.

superm1, 2018-01-11:
Also, FYI, filed this https://github.com/rhboot/efivar/issues/96 to request new efivar release.

ArchangeGabriel, 2018-01-11:
Thanks, I’m looking into getting the patch in ArchLinux in the meantime.

superm1, 2018-01-11:
Cool.  I'm making it a little more explicit that efivar 33 is needed in master.  It should be released before the next tagged version.

closed 2018-01-11 by superm1

ArchangeGabriel, 2018-01-11:
Right, but it’s not just master, it’s also in fwupdate-10 tarball.

superm1, 2018-01-11:
Yeah I know, not much that can be done there.  Hopefully @vathpela can tag a new efivar release soon.  I've made it clear in the README that efivar 33 is needed.

commit b9b83ab8 (2018-01-11): require efivar 33 (Closes #93)