rhboot/fwupdate #95: Status for Lenovo hardware

t-8ch, 2018-01-28:
Is there any public documentation on how to update a Lenovo (Thinkpad) UEFI with fwupdate?

The vendorlist show it as supported.
Also my Firmware update release notes mention fixes in the WUFU compatability.
Furthermore the update package from Lenovo itself contains strings that indicate that some sort of support should exist:

```
Found fwupdate variable, set as Linux WUFU (fwupd) triggered capsule update.
Delete fwupdate variable successfully.
```

However I could not find any public documentation about this.

Sorry if this was the wrong place to ask.

hughsie, 2018-01-29:
Perhaps answering the question the reporter was asking: Lenovo are testing the LVFS with some firmware, but it's limited in scope and doesn't include ThinkPads yet. Fingers crossed.

t-8ch, 2018-01-29:
I was trying to extract or generate a capsule from the normal BIOS update .exe/.iso and stumbled upon these strings.
Thanks for the clarification.

superm1, 2018-01-29:
Assuming Lenovo is publishing BIOS updates to Windows Update for their Thinkpads, you might be able to find UEFI capsules on the WU catalog: https://www.catalog.update.microsoft.com.

hughsie, 2018-01-29:
I can't say why, but I'm going to wait for Lenovo to issue updates specifically for fwupdate.

t-8ch, 2018-01-29:
@superm1 This looks like it contains a Capsule for my device, thanks!

@hughsie 
"You can't say why" because you don't know how to express it, or for legal reasons?

hughsie, 2018-01-29:
Both I guess :) Let me know if it works, but if you try use fwupd from master and turn on debugging. R

superm1, 2018-01-29:
(Disclaimer - I have no knowledge of where the problems @hughsie is referring to are).

If you test directly from the CAB file you would be testing from fwupd (which would use libfwup from fwupdate to do the staging).  

If you extract the CAB file and just grab the payload you can test directly from fwupdate (which also uses libfwup, just a different interface).

If I was going to take a guess from the comments in this thread the problems probably aren't in fwupd, but they're probably in the fwupdate EFI application with what Lenovo's BIOS expects.

So it would be best to use the "10" release (or master if you can swing it - there's extra debugging now) of fwupdate if there are some issues with capsule to have the highest liklihood of success and capability to debug if that's where the problems end up being.

t-8ch, 2018-01-29:
The debug log looks fine, but the updates has not been performed (debug log attached).
Can I be of any assistance in debugging this, or am I just annoying and should wait until the vendor provides support?
Also I will be at FOSDEM, so maybe this could help.

```
fwupdate.c:463:find_updates(): Found update fwupdate-42f29620-3d63-4f09-950b-9b7055570f28-0
fwupdate.c:1290:efi_main(): Adding new capsule
fwupdate.c:574:search_file(): Searching Device Path: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)/\\\\\\EFI\arch\fw\fwupdate-dQRgOb.cap ...
fwupdate.c:602:search_file(): Device Path prepared: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:614:search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:621:search_file(): Comparing: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158) and PciRoot(0)/Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:621:search_file(): Comparing: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158) and Pci(0x17,0x0)/Sata(0x1,0x0,0x0)/HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:621:search_file(): Comparing: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158) and Sata(0x1,0x0,0x0)/HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:621:search_file(): Comparing: HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158) and HD(Part1,SigFA4545EC-5EDC-4B42-923F-8969E059D158)
fwupdate.c:631:search_file(): Match up! Returning \\\\\\EFI\arch\fw\fwupdate-dQRgOb.cap
fwupdate.c:642:search_file(): File \\\\\\EFI\arch\fw\fwupdate-dQRgOb.cap searched
fwupdate.c:703:open_file(): Found device
fwupdate.c:710:open_file(): Found volume
fwupdate.c:718:open_file(): Found file
fwupdate.c:1050:add_capsule(): Read file; 108 bytes
fwupdate.c:1051:add_capsule(): updates guid: 42F29620-3D63-4F09-950B-9B7055570F28
fwupdate.c:1052:add_capsule(): File guid: 0000006C-0000-0000-10E7-480000000000
fwupdate.c:1086:add_capsule(): Image does not have embedded header
fwupdate.c:1088:add_capsule(): Allocating 136 for capsule header.
fwupdate.c:1304:efi_main(): n_updates: 1
fwupdate.c:1154:apply_capsules(): QueryCapsuleCapabilities: Success max: 342552576 reset:1
fwupdate.c:1155:apply_capsules(): Capsules: 1
```

superm1, 2018-01-29:
Nothing useful stands out in that log to me.  It's basically saying the payload was handed off to the firmware (presumably properly?)

After you come back to the OS if you query the results from fwupdate (`fwupdate --info` or `fwupdmgr get-results`) that might give some hints that the firmware dropped in the ESRT about what happened.

If there's nothing useful there, it probably needs a debug BIOS from the vendor to find out what's going wrong.

t-8ch, 2018-01-29:
```
# fwupdate --info
Disabled fwupdate debugging

Information for the update status entry 0:
  Information Version: 7
  Firmware GUID: {ffaf3e12-a6ea-4352-851b-76c6455913c6}
  Capsule Flags: 0x00000000
  Hardware Instance: 0
  Update Status: Unknown
  Capsule File Path: 

Information for the update status entry 1:
  Information Version: 7
  Firmware GUID: {42f29620-3d63-4f09-950b-9b7055570f28}
  Capsule Flags: 0x00000000
  Hardware Instance: 0
  Update Status: Attempted
  Attempted Time: Mon Jan 29 19:34:42 2018
  Capsule File Path: HD(1,GPT,fa4545ec-5edc-4b42-923f-8969e059d158,0x800,0x100000)/File(\\\\\\EFI\arch\fw\fwupdate-dQRgOb.cap)
```

superm1, 2018-01-29:
Well that's notable that there are two ESRT entries.  It looks like the second represents your system firmware but I'm not sure what the first one represented. 

I suppose that could also cause some display issues with fwupd output.  I haven't ran fwupd on something with two separate entries in ESRT.  Does the `# fwupdmgr get-devices `output look normal to you?  (that would be a separate problem)

t-8ch, 2018-01-29:
```
# fwupdmgr get-devices
UEFI Device Firmware
  DeviceId:             33bc73bf0f48eb6177a007e9351e22d5bce2c04d
  Guid:                 ffaf3e12-a6ea-4352-851b-76c6455913c6
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.11
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2018-01-29

20FAS05Q00 System Firmware
  DeviceId:             444a229da2b21bc0d24faf814d1c58c5bff3315f
  Guid:                 42f29620-3d63-4f09-950b-9b7055570f28
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.25
  VersionLowest:        0.1.22
  Icon:                 computer
  Created:              2018-01-29

Skylake GT2 [HD Graphics 520]
  DeviceId:             ca9f93cb97772027f45ba73c86cf4980fb02745b
  Guid:                 a3361e1f-f523-5cfc-ad68-e20d0f792003
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-01-29
```

Maybe the information from dmidecode helps:
```
BIOS Information
        Vendor: LENOVO
        Version: N1CET57W (1.25 )
        Release Date: 05/25/2017
        Address: 0xE0000
        Runtime Size: 128 kB
        ROM Size: 16 MB
        Characteristics:
                PCI is supported
                PNP is supported
                BIOS is upgradeable
                BIOS shadowing is allowed
                Boot from CD is supported
                Selectable boot is supported
                EDD is supported
                3.5"/720 kB floppy services are supported (int 13h)
                Print screen service is supported (int 5h)
                8042 keyboard services are supported (int 9h)
                Serial services are supported (int 14h)
                Printer services are supported (int 17h)
                CGA/mono video services are supported (int 10h)
                ACPI is supported
                USB legacy is supported
                BIOS boot specification is supported
                Targeted content distribution is supported
                UEFI is supported
        BIOS Revision: 1.25
        Firmware Revision: 1.11
```

superm1, 2018-01-29:
Oh it came up as a device firmware not system. OK.
Does 
`# fwupdmgr get-results 444a229da2b21bc0d24faf814d1c58c5bff3315f
`
Say anything else that wasn't already obtained in `fwupdate --info`?

t-8ch, 2018-01-29:
```
UpdateState:          success
```

superm1, 2018-01-29:
OK, then I don't know what more can be done here without Lenovo's help.

t-8ch, 2018-01-29:
Ok, I will wait patiently :-)
Thanks for the help!

hughsie, 2018-01-29:
>If there's nothing useful there, it probably needs a debug BIOS from the vendor

This is the conclusion I also reached.

t-8ch, 2018-08-06:
I assume this can be closed with a "Thank you!"

closed 2018-08-06 by t-8ch