rhboot/fwupdate #101: Hardening breaks build

jtojnar, 2018-02-21:
On NixOS, we use hardening by default and unless we disable `-fstack-protector-strong --param ssp-buffer-size=4`, the build will fail with:

```
make[1]: Entering directory '/build/fwupdate-10/efi'
gcc -Og -g3 -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -fpic -Werror -Wall -Wextra -fshort-wchar -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/build/fwupdate-10/include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/nix/store/0081d66ckxlrsyszlxhjbmcn3z11wxcd-gcc-6.4.0/lib/gcc/x86_64-unknown-linux-gnu/6.4.0/include -c -o fakeesrt2.o fakeesrt2.c
gcc -nostdlib -Wl,--warn-common -Wl,--no-undefined -Wl,--fatal-warnings -Wl,-shared -Wl,-Bsymbolic -L/nix/store/0r8wbqnswd73zi404qribx56sl6y6vag-fwupdate-10/lib -L/nix/store/vs2691kx6grcn3qwv97xibdj1mlj1cbj-gnu-efi-3.0.6/lib -Wl,--build-id=sha1 -Wl,--hash-style=sysv /nix/store/vs2691kx6grcn3qwv97xibdj1mlj1cbj-gnu-efi-3.0.6/lib/crt0-efi-x86_64.o -o fakeesrt2.so fakeesrt2.o -lefi -lgnuefi \
        /nix/store/0081d66ckxlrsyszlxhjbmcn3z11wxcd-gcc-6.4.0/lib/gcc/x86_64-unknown-linux-gnu/6.4.0/libgcc.a \
        -T elf_x86_64_efi.lds
fakeesrt2.o: In function `efi_main':
/build/fwupdate-10/efi/fakeesrt2.c:56: undefined reference to `__stack_chk_fail'
collect2: error: ld returned 1 exit status
make[1]: *** [/build/fwupdate-10/efi/Makefile:107: fakeesrt2.so] Error 1
rm fakeesrt2.o
make[1]: Leaving directory '/build/fwupdate-10/efi'
make: *** [Makefile:15: all] Error 2
```

superm1, 2018-02-21:
What compiler is this?

Are you sure those are what's causing it?  They're also in my build env and don't fail.

```
root@5314b8878ad4:/build/efi# make GNUEFIDIR=/usr/lib
gcc -Og -g3 -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -fpic -Werror -Wall -Wextra -fshort-wchar -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/build/include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/usr/lib/gcc/x86_64-linux-gnu/7/include -c -o fakeesrt2.o fakeesrt2.c                                                          
gcc -nostdlib -Wl,--warn-common -Wl,--no-undefined -Wl,--fatal-warnings -Wl,-shared -Wl,-Bsymbolic -L/usr/lib64 -L/usr/lib -Wl,--build-id=sha1 -Wl,--hash-style=sysv /usr/lib/crt0-efi-x86_64.o -o fakeesrt2.so fakeesrt2.o -lefi -lgnuefi \                                                                                                                                                                                  
        /usr/lib/gcc/x86_64-linux-gnu/7/libgcc.a \
        -T elf_x86_64_efi.lds
objcopy -j .text -j .sdata -j .data -j .dynamic -j .dynsym \
        -j .rel* -j .rela* -j .reloc -j .eh_frame \
        --target efi-app-x86_64 fakeesrt2.so fakeesrt2.efi
rm fakeesrt2.o fakeesrt2.so
```

```
root@5314b8878ad4:/build/efi# gcc --version | head -1
gcc (Ubuntu 7.3.0-3ubuntu1) 7.3.0
```

jtojnar, 2018-02-22:
We use `gcc (GCC) 6.4.0`. And I think it is about order of arguments: In `efi/Makefile` it is first turned on with `-fstack-protector-strong`, then disabled using `-fno-stack-protector` and finally re-enabled by Nix. I guess we should prevent enabling it for the `efi` directory.

closed 2018-02-22 by jtojnar