rhboot/fwupdate #115: fwupdate doesn't see Dell XPS 13 bios update

tapir, 2018-06-04:
dmidecode gives this
```
Handle 0x0000, DMI type 0, 24 bytes
BIOS Information
	Vendor: Dell Inc.
	Version: 2.5.0
	Release Date: 12/12/2017
	Address: 0xF0000
	Runtime Size: 64 kB
	ROM Size: 16 MB
	Characteristics:
		PCI is supported
		PNP is supported
		BIOS is upgradeable
		BIOS shadowing is allowed
		Boot from CD is supported
		Selectable boot is supported
		EDD is supported
		Japanese floppy for NEC 9800 1.2 MB is supported (int 13h)
		5.25"/1.2 MB floppy services are supported (int 13h)
		3.5"/720 kB floppy services are supported (int 13h)
		3.5"/2.88 MB floppy services are supported (int 13h)
		Print screen service is supported (int 5h)
		8042 keyboard services are supported (int 9h)
		Serial services are supported (int 14h)
		Printer services are supported (int 17h)
		ACPI is supported
		USB legacy is supported
		Smart battery is supported
		BIOS boot specification is supported
		Function key-initiated network boot is supported
		Targeted content distribution is supported
		UEFI is supported
	BIOS Revision: 2.5
```
and fwupdmgr get-devices gives this
```
Intel AMT [unprovisioned]
  DeviceId:             088df415cdee883ec89563e41e6d495924250174
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              11.8.50
  VersionBootloader:    11.8.50
  Icon:                 computer
  Created:              2018-06-04

XPS 13 9360 TPM 2.0
  DeviceId:             3368246a16d77193e42a2fea8d95439b38f01670
  Guid:                 2e513587-0d9b-50e6-8e15-1845d44e88eb
  Summary:              Platform TPM device
  Plugin:               dell
  Flags:                internal|require-ac|registered
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2018-06-04

UHD Graphics 620
  DeviceId:             8de6c7959053fd5798006dcc63590d33fa5e51cb
  Guid:                 8eb8bd2e-0fca-5aba-9aa8-f341e0aa4482
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2018-06-04
```

I see that there is a BIOS update 2.6.2 which is not recognized buy fwupdmgr

tapir, 2018-06-04:
Sorry it's probably related to this
```
[josh@collin Downloads]$ fwupdate --supported
Firmware updates are not supported on this machine.
[josh@collin Downloads]$ sudo  fwupdate --enable
Firmware updates can not be enabled on this machine from this tool.
```

Any way to enable it?

superm1, 2018-06-04:
Which model XPS is this?  You may need to go into BIOS setup to turn on UEFI capsule updates if it's not already on.

tapir, 2018-06-04:
I thought it was an XPS 9365 because it's the 2018 model but not the newest design (old case) but looks like it's 9360 from what I see on system messages

superm1, 2018-06-04:
No that shouldn't make any difference.

What OS are you running?  Do you have ESRT enabled in your kernel?

tapir, 2018-06-04:
I'm using arch linux
```
[cosku@collin ~]$ lsmod | grep esrt
[cosku@collin ~]$ lsmod | grep ESRT
[cosku@collin ~]$ 
```
looks like I'm not, is it just a matter of doing 
```
modprobe esrt
```
?

superm1, 2018-06-04:
Can you please check your kernel config?  You should have `CONFIG_EFI_ESRT` set to something.

You can also check if you have a `/sys/firmware/efi/esrt` directory.  If you don't have that, that would explain fwupdate's lack of working.

ArchangeGabriel, 2018-06-04:
Are you booting in EFI mode at all?

tapir, 2018-06-04:
Good question :) How can I check that?

ArchangeGabriel, 2018-06-04:
In your BIOS, Boot Options or something like that. But the absence of `/sys/firmware/efi` is generally a good indicator that’s not the case. Also, you should by the way you installed Arch, especially your bootloader.

superm1, 2018-06-04:
Agreed, it really looks to me that this isn't UEFI mode and that fully explains all of this behavior.

closed 2018-06-04 by superm1

tapir, 2018-06-04:
I'm actually using Antergos so the installation was automatic. Thanks for diagnostics!

ArchangeGabriel, 2018-06-04:
There might also be an already existing ESP depending on how the installation happened. Assuming there was some OEM system installed in the first place, there likely was an ESP at that time, and it might still exist. The output of `fdisk -l` could be interesting in that purpose (as well to check about GPT).

tapir, 2018-06-04:
```
sudo parted -l
Partition Table: msdos```

In that case I think I'll just use the F12 - USB BIOS Flash instead
Thank you both