rhboot/fwupdate #118: Make fwupdate.efi able to set BootNext

vathpela, 2018-06-25:
Dunno if it'll actually work, but spec-wise it looks like it's supposed to, and it'll help the beaker case.

superm1, 2018-07-09:
I guess I don't know the situation you're referring to when you say "beaker case", but why would `BootNext` ever need to be set in the EFI application?  When you call `UpdateCapsule` FW will have taken over (potentially power cycling) and after FW update is done should just resume it's normal boot path to `BootOrder`

If this is for some custom type PXE solution, I think the easier approach it is to just set `BootOrder` to have whatever is supposed to come next after the FW update is done.

enunes, 2018-07-10:
That would be for automated testing with Beaker.

Beaker (https://beaker-project.org/) requires machines to be set up with PXE always first, so that it can perform the OS installation before running jobs. Since installers are always changing BootOrder around, unfortunately in practice it is not possible to rely on it finding a disk/OS after that, often it ends up dropping to EFI shell, etc. Therefore Beaker relies on setting BootNext to make sure jobs in UEFI machines won't abort during reboots (https://beaker-project.org/docs/user-guide/task-environment.html#rhts-reboot).

fwupdate also relies on BootNext, and this is inconvenient in order to have automated fwupdate tasks. It requires the automated task to change BootOrder like you suggested, but that breaks the requirement above in case something goes wrong and leaves a broken automated machine.

@vathpela suggested that it might be possible for fwupdate to support a flag that will set a boot entry to BootNext after running the firmware update attempt, which would useful in this case.

superm1, 2018-07-10:
I guess I follow the logic you're mentioning here.  It's basically you want to create a `BootNextNext` variable that `fwupdate.efi` will rename to `BootNext` when it's running.

I think it will still be awkward though because you are calling `rhts-reboot` as a last step right?  And that will be what's setting `BootNext` to come to the OS again.  So you probably need to have something changing both there and in `fwupdate` in concert to make this act like you want it to.

superm1, 2020-04-09:
https://github.com/rhboot/fwupdate/pull/136

closed 2020-04-09 by superm1