rhboot/fwupdate #123: Compilation failed on Pop!_OS 18.04

aral, 2018-10-16:
```bash
 make
Building with EFIDIR as ubuntu
make[1]: Entering directory '/home/aral/sandbox/fwupdate/efi'
gcc -Og -g3 -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -fpic -Werror -Wall -Wextra -fshort-wchar -Wno-error=missing-field-initializers -Wno-missing-field-initializers -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/home/aral/sandbox/fwupdate/include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/usr/lib/gcc/x86_64-linux-gnu/7/include -c -o fakeesrt2.o fakeesrt2.c
fakeesrt2.c:1:10: fatal error: efi.h: No such file or directory
 #include <efi.h>
          ^~~~~~~
compilation terminated.
/home/aral/sandbox/fwupdate/efi/Makefile:119: recipe for target 'fakeesrt2.o' failed
make[1]: *** [fakeesrt2.o] Error 1
make[1]: Leaving directory '/home/aral/sandbox/fwupdate/efi'
Makefile:16: recipe for target 'all' failed
make: *** [all] Error 2
```

aral, 2018-10-16:
I also get the feeling that even if it compiled correctly, I'd still not be able to get this working under Pop!_OS with my Dell XPS 13 due to:

```bash
 sudo fwupdmgr update 
[sudo] password for aral: 
Downloading 0.1.5.1 for XPS 13 9370 System Firmware...
Updating 0.1.5.1 on XPS 13 9370 System Firmware...
Decompressing…         [***************************************]
Authenticating…        [***************************************]
Scheduling…            [***************************************]
UEFI firmware update failed: {error #0} libfwup.c:1501 get_fd_and_media_path(): failed to make /boot/efi/EFI/ubuntu/fw: No such file or directory
```

Listings of the relevant directories:

```bash
  sudo ls /boot/efi/EFI/
BOOT  Pop_OS-83675915-83e5-4284-a68a-261d82931428  systemd
Dell  Recovery-7354-E439
  sudo ls /boot/efi/EFI/Pop_OS-83675915-83e5-4284-a68a-261d82931428
cmdline  initrd.img  initrd.img-previous  vmlinuz.efi  vmlinuz-previous.efi
```

(There is no ubuntu directory, even though the system specifies its ID as such and in the Pop_OS... directory, I can't see a an _fw_ entry even if the directory were correct.)

superm1, 2018-10-16:
Do you have `gnu-efi` installed?  It seems like no.

superm1, 2018-10-16:
But honestly; this project is on life support and just a reference implementation now.  If there is a bug with derivative handling it's probably not going to be fixed.

it would be better to do this with fwupd 1.1.x (which has subsumed this project).

Do you have support for snap installed on PopOS?  Or can you?  If so, can you please try this:

```
sudo apt purge fwupd
sudo snap install --candidate --classic fwupd
fwupdgmr refresh
fwupdmgr update
```

If that still fails, we might have some derivative handling problems on the fwupd UEFI implementation.

aral, 2018-10-16:
@superm1 Thanks, I do now :) And `make` got a bit further... is there a list of dependencies anywhere so I don't have to keep bothering you? :)

(Ah, and just saw your other reply - I do and thank you, I will try that and report back. I was rather confused about the difference between fwupdate, fwupd, and fwupdmgr. Good to know that fwupd is the one to go with.)

PS. The latest make error:

```bash
make[1]: Entering directory '/home/aral/sandbox/fwupdate/linux'
Package efivar was not found in the pkg-config search path.
Perhaps you should add the directory containing `efivar.pc'
to the PKG_CONFIG_PATH environment variable
No package 'efivar' found
Package efiboot was not found in the pkg-config search path.
Perhaps you should add the directory containing `efiboot.pc'
to the PKG_CONFIG_PATH environment variable
No package 'efiboot' found
if [ ! -d include ]; then \
	install -d -m 755 include ; \
fi
sed -e "s,@@FWUP_VERSION@@,12,g" \
	-e "s,@@FWUP_MAJOR_VERSION@@,,g" \
	-e "s,@@FWUP_MINOR_VERSION@@,,g" \
	-e "s,@@PREFIX@@,/usr,g" \
	-e "s,@@EXEC_PREFIX@@,/usr,g" \
	-e "s,@@SHAREDSTATEDIR@@,/var/lib,g" \
	-e "s,@@ESPMOUNTPOINT@@,"/boot/efi",g" \
	-e "s,@@EFIDIR@@,ubuntu,g" \
	-e "s,@@LIBDIR@@,/usr/lib/x86_64-linux-gnu,g" \
	-e "s,@@LIBEXECDIR@@,/usr/lib,g" \
	-e "s,@@INCLUDEDIR@@,/usr/include,g" \
	include/fwup-version.h.in > include/fwup-version.h
sed -e "s,@@FWUP_VERSION@@,12,g" \
	-e "s,@@FWUP_MAJOR_VERSION@@,,g" \
	-e "s,@@FWUP_MINOR_VERSION@@,,g" \
	-e "s,@@PREFIX@@,/usr,g" \
	-e "s,@@EXEC_PREFIX@@,/usr,g" \
	-e "s,@@SHAREDSTATEDIR@@,/var/lib,g" \
	-e "s,@@ESPMOUNTPOINT@@,"/boot/efi",g" \
	-e "s,@@EFIDIR@@,ubuntu,g" \
	-e "s,@@LIBDIR@@,/usr/lib/x86_64-linux-gnu,g" \
	-e "s,@@LIBEXECDIR@@,/usr/lib,g" \
	-e "s,@@INCLUDEDIR@@,/usr/include,g" \
	libfwup.map.in > libfwup.map
gcc -g3 -Og -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -Wall -Wextra -Werror -Wno-error=cpp -Wno-unused-result -Wno-unused-function -Wsign-compare -Werror=sign-compare -Wno-error=missing-field-initializers -Wno-missing-field-initializers -fshort-wchar --std=gnu11 -DLOCALEDIR=\"/usr/share/locale\" -D_GNU_SOURCE -DFWUP_EFI_DIR_NAME=\"ubuntu\" -DFWUP_ESP_MOUNTPOINT=\""/boot/efi"\" -I include -I /home/aral/sandbox/fwupdate/linux/include/ -iquote /home/aral/sandbox/fwupdate/include/      -fPIC -c -o libfwup.o libfwup.c
libfwup.c:14:10: fatal error: efivar/efiboot.h: No such file or directory
 #include <efivar/efiboot.h>
          ^~~~~~~~~~~~~~~~~~
compilation terminated.
/home/aral/sandbox/fwupdate/linux/Makefile:100: recipe for target 'libfwup.o' failed
make[1]: *** [libfwup.o] Error 1
make[1]: Leaving directory '/home/aral/sandbox/fwupdate/linux'
Makefile:16: recipe for target 'all' failed
make: *** [all] Error 2
```

superm1, 2018-10-16:
>  was rather confused about the difference between fwupdate, fwupd, and fwupdmgr

Not sure if you saw https://github.com/rhboot/fwupdate/blob/master/README.md.  I was hoping that made it clear.

You next error with fwupdate means you're missing the other dependency efivar/efiboot (`libefivar-dev` and `libefiboot-dev`).

But yeah I do think go down the fwupd path.  If you have problems with fwupd still file them here https://github.com/hughsie/fwupd/issues and we'll debug 'em.

aral, 2018-10-16:
@superm1 Well, seeing and reading are apparently two different things :man_facepalming:

Thank you so much for the _fwupd_ tip. That did it and I'm up and running on the latest firmware. I'm going to write up a quick blog post so that anyone hitting the same issue with Pop!_OS on an XPS can find it easily.

Lots of love from the Emerald Isle :)

closed 2018-10-16 by aral

superm1, 2018-10-16:
Great, thanks!  Link it back here so that you complete the loop.

aral, 2018-10-16:
Done - thanks again for your help :)

https://ar.al/2018/10/16/updating-firmware-on-dell-xps-13-with-popos-18.04/

kendocode, 2019-02-02:
Same solution for same problem worked on my p71