rhboot/fwupdate #125: how to build this modules

MarvinLian, 2018-12-08:
Hi Team,
I try to build this module using the command "make -C /lib/modules/4.13.0-46-generic/build M=$(pwd)"
and the result as below, could you help to reply how to build this module.
```
make: Entering directory '/usr/src/linux-headers-4.13.0-46-generic'
  Building modules, stage 2.
  MODPOST 0 modules
make: Leaving directory '/usr/src/linux-headers-4.13.0-46-generic'
```
my gcc version is "gcc (Ubuntu 7.2.0-8ubuntu3.2) 7.2.0"
my make version is "GNU Make `4.1"`

superm1, 2018-12-08:
This isn't a kernel module, you shouldn't build with a kernel build command.

superm1, 2018-12-08:
See https://github.com/rhboot/fwupdate/blob/master/README.md for compilation instructions

closed 2018-12-08 by superm1

MarvinLian, 2018-12-10:
Hi Team,
Actually, I try to build this module as a normal application. And I had installed  libpopt-dev (version 1.16-11)  efivar (version 34), gnu-efi (version 3.0.8), elfutils (0.170-0.5.0). execute make command, I got the below error, could you help check why.
```
Building with EFIDIR as ubuntu
make[1]: Entering directory '/home/ert/fwupdate/efi'
make[1]: Nothing to be done for 'all'.
make[1]: Leaving directory '/home/ert/fwupdate/efi'
make[1]: Entering directory '/home/ert/fwupdate/linux'
/bin/sh: 1: pkg-config: not found
make[1]: pkg-config: Command not found
make[1]: pkg-config: Command not found
if [ ! -d include ]; then \
	install -d -m 755 include ; \
fi
gcc -g3 -Og -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -Wall -Wextra -Werror -Wno-error=cpp -Wno-unused-result -Wno-unused-function -Wsign-compare -Werror=sign-compare -Wno-error=missing-field-initializers -Wno-missing-field-initializers -fshort-wchar --std=gnu11 -DLOCALEDIR=\"/usr/share/locale\" -D_GNU_SOURCE -DFWUP_EFI_DIR_NAME=\"ubuntu\" -DFWUP_ESP_MOUNTPOINT=\""/boot/efi"\" -I include -I /home/ert/fwupdate/linux/include/ -iquote /home/ert/fwupdate/include/      -fPIC -c -o libfwup.o libfwup.c
In file included from libfwup.c:31:
include/fwup.h:15:10: fatal error: efivar.h: No such file or directory
 #include <efivar.h>
          ^~~~~~~~~~
compilation terminated.

```