rhboot/fwupdate #128: New warning with GCC 9

marxin, 2019-01-09:
Similar to https://github.com/rhboot/efivar/issues/123: 

```
[   19s] In file included from /usr/include/efi/efi.h:41,
[   19s]                  from fwupdate.c:11:
[   19s] fwupdate.c: In function 'find_updates':
[   19s] fwupdate.c:529:8: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]   529 |        &update->info->time_attempted,
[   19s]       |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[   19s] /usr/include/efi/x86_64/efibind.h:296:51: note: in definition of macro 'uefi_call_wrapper'
[   19s]   296 | #define uefi_call_wrapper(func, va_num, ...) func(__VA_ARGS__)
[   19s]       |                                                   ^~~~~~~~~~~
[   19s] fwupdate.c: In function 'add_capsule':
[   19s] fwupdate.c:1060:16: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]  1060 |  if ((guid_cmp(&update->info->guid, (efi_guid_t *)fbuf) == 0 ||
[   19s]       |                ^~~~~~~~~~~~~~~~~~~
[   19s] fwupdate.c:1076:41: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]  1076 |   if (!cap_out->Flags && !is_ux_capsule(&update->info->guid)) {
[   19s]       |                                         ^~~~~~~~~~~~~~~~~~~
[   19s] fwupdate.c:1023:39: note: in definition of macro 'is_ux_capsule'
[   19s]  1023 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   19s]       |                                       ^~~~
[   19s] fwupdate.c:1099:22: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]  1099 |   if (!is_ux_capsule(&update->info->guid)) {
[   19s]       |                      ^~~~~~~~~~~~~~~~~~~
[   19s] fwupdate.c:1023:39: note: in definition of macro 'is_ux_capsule'
[   19s]  1023 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   19s]       |                                       ^~~~
[   19s] fwupdate.c:1118:20: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]  1118 |  if (is_ux_capsule(&update->info->guid)) {
[   19s]       |                    ^~~~~~~~~~~~~~~~~~~
[   19s] fwupdate.c:1023:39: note: in definition of macro 'is_ux_capsule'
[   19s]  1023 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   19s]       |                                       ^~~~
[   19s] fwupdate.c: In function 'efi_main':
[   19s] fwupdate.c:1295:22: error: taking address of packed member of 'struct update_info_s' may result in an unaligned pointer value [-Werror=address-of-packed-member]
[   19s]  1295 |        is_ux_capsule(&updates[i]->info->guid))
[   19s]       |                      ^~~~~~~~~~~~~~~~~~~~~~~
[   19s] fwupdate.c:1023:39: note: in definition of macro 'is_ux_capsule'
[   19s]  1023 | #define is_ux_capsule(guid) (guid_cmp(guid, &ux_capsule_guid) == 0)
[   19s]       |                                       ^~~~
```

marxin, 2019-01-09:
Compiler options:

```
[   19s] gcc -Og -g3 -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -fexceptions -fstack-protector-strong --param=ssp-buffer-size=4 -grecord-gcc-switches -fpic -Werror -Wall -Wextra -fshort-wchar -Wno-error=missing-field-initializers -Wno-missing-field-initializers -fno-merge-constants -ffreestanding -fno-stack-protector -fno-stack-check --std=gnu11 -DCONFIG_x86_64 -I/usr/include/efi/ -I/usr/include/efi/x86_64/ -iquote/home/abuild/rpmbuild/BUILD/fwupdate-12/include "-DDEBUGDIR=L\"/\"" -mno-mmx -mno-sse -mno-red-zone -nostdinc -maccumulate-outgoing-args -DEFI_FUNCTION_WRAPPER -DGNU_EFI_USE_MS_ABI -I/usr/lib64/gcc/x86_64-suse-linux/9/include -c -o fwupdate.o fwupdate.c
```

marxin, 2019-03-08:
May I please ping this?

hughsie, 2019-03-08:
Out of interest, is there any reason you're using fwupdate and not fwupd?

marxin, 2019-03-08:
> Out of interest, is there any reason you're using fwupdate and not fwupd?

Huh, I'm just aware that we both these package in our openSUSE distribution. Is any of these in a legacy mode?

hughsie, 2019-03-08:
> any of these in a legacy mode?

fwupd is completely replacing fwupdate. See https://blogs.gnome.org/hughsie/2018/07/02/fwupdate-is-nearly-dead-long-live-fwupd/

marxin, 2019-03-08:
> fwupd is completely replacing fwupdate. See https://blogs.gnome.org/hughsie/2018/07/02/fwupdate-is-nearly-dead-long-live-fwupd/

Thanks for information!
Should I close this issue then?

hughsie, 2019-03-08:
I think so.

closed 2019-03-08 by hughsie