rhboot/fwupdate #130: arm64 version of PE/COFF efi binary is corrupt

superm1, 2019-04-09:
This issue was reported in Debian with fwupdate 12.

> Version 12-4 of fwupdate is broken for arm64. The included binary fwupaa64.efi
> is corrupt, resulting in EFI_LOAD_ERROR to be returned by the firmware when
> trying to invoke it.

The binary layout looks like this: 

```
Detected 'AArch64' type PE/COFF image consisting of 2 sections
Section alignment: 0x1000
File alignment: 0x200
Image size: 0xd890
Section '.text' @ 0x00001000
File offset: 0x1000
Virtual size: 0xac20
Raw size: 0xac20
Section '.data' @ 0x0000bc20
File offset: 0xbc20
Virtual size: 0x1d70
Raw size: 0x1d70
```

> Note that file offset + size of section #2 exceeds the total image size. But
> the file offset of that section is not even a multiple of the file alignment,
> so the whole image seems pretty broken.
>

superm1, 2019-04-09:
Issue forwarded from: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=926596

nhivp, 2019-12-24:
Hi @superm1 ,
This issue is caused by gnu-efi library. After checking the gnu-efi source code, I see the commit https://sourceforge.net/p/gnu-efi/code/ci/8581a58e5ba8730cda24fd1ad550653c74059f3d/ cause this issue.

superm1, 2019-12-25:
Thanks! Can you please open the PR with the solution in fwupd project too?

nhivp, 2019-12-26:
The main issue is that the size of .data section has calculated incorrectly. The starting point of .data section should be at _data, not _etext.

The PR #134 already fixed this issue. Please merge it.

superm1, 2020-01-06:
@nhivp Thanks it's been merged.  As I can tell, a similar change for fwupd is not necessary because fwupd takes the lds file distributed with gnu-efi: https://github.com/fwupd/fwupd/blob/master/plugins/uefi/efi/meson.build#L33

If you can confirm it works properly with fwupd, it would be appreciated!

closed 2020-01-06 by superm1

nhivp, 2020-01-13:
Hi @superm1,

I installed the fwupd on Ubuntu 20.04 LTS (Focal Fossa) for arm64.
```
root@ubuntu:~# fwupdmgr --version
client version: 1.3.6
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.3.6
```

This issue doesn't happen. I can update a UEFI system firmware successfully.
Furthermore, you can check the following pev output:
```
$ pev fwupdaa64.efi
DOS header:
 Magic number:			0x5a4d (MZ)
 Bytes in last page:		0
 Pages in file:			0
 Relocations:			0
 Size of header in paragraphs:	0
 Minimum extra paragraphs:	0
 Maximum extra paragraphs:	0
 Initial (relative) SS value:	0
 Initial SP value:		0
 Initial IP value:		0
 Initial (relative) CS value:	0
 Address of relocation table:	0
 Overlay number:		0
 OEM identifier:		0
 OEM information:		0
 PE header offset:		0x40

COFF header:
 Machine:			0xaa64 - Unknown machine type
 Number of sections:		2
 Date/time stamp:		0 (Thu, 01 Jan 1970 00:00:00 UTC)
 Symbol Table offset:		0
 Number of symbols:		1
 Size of optional header:	0xa0
 Characteristics:		0x206 (0000001000000110)
				executable image
				line numbers removed (deprecated)
				debugging information removed

Optional (PE) header:
 Magic number:			0x20b (PE32+)
 Linker major version:		2
 Linker minor version:		20
 Size of .code section:		0xa000
 Size of .data section:		0x1c00
 Size of .bss section:		0
 Entry point:			0x1000
 Address of .code section:	0x1000
 Imagebase:			0
 Alignment of sections:		0x1000
 Alignment factor:		512
 Major version of required OS:	0
 Minor version of required OS:	0
 Major version of image:	0
 Minor version of image:	0
 Major version of subsystem:	0
 Minor version of subsystem:	0
 Size of image:			0xcc00
 Size of headers:		0x1000
 Image file checksum:		0
 Subsystem required:		10 (XBOX)
 DLL characteristics:		0 (0000000000000000)
 Size of stack to reserve:	0
 Size of stack to commit:	0
 Size of heap space to reserve:	0
 Size of heap space to commit:	0
 Data-dictionary entries:	6

Data directories:
Sections:
 Name:				.text
 Virtual size:			0xa000
 Virtual address:		0x1000
 Data size:			0xa000
 Data offset:			0x1000
 Characteristics:		0x60000020 (01100000000000000000000000100000)
				contains executable code
				is executable
				is readable

 Name:				.data
 Virtual size:			0x1c00
 Virtual address:		0xb000
 Data size:			0x1c00
 Data offset:			0xb000
 Characteristics:		0xc0000040 (11000000000000000000000001000000)
				contains initialized data
				is readable
				is writable

pev: the file does not contain a valid resource section
```

As you can see from the output above, the sum of _Data size_ and _Data offset_ is not over the _Size of image_.

superm1, 2020-01-13:
Thanks for confirming!