rhboot/fwupdate #133: nothing happens after reboot

emilas44, 2019-08-17:
I have a X1 Extreme and fwupdate does nothing after reboot.... if I run fwupdmgr update it downloads the update file and it asks if I want to reboot and If I type in Y and Enter it will reboot and comes normally up to login screen and does nothing. I even made a video so you guys can see it. Maybe someone can suggest something what can I try or do to make it work?

https://www.youtube.com/watch?v=awlPrj4YX0w

hughsie, 2019-08-18:
I think you want the fwupd project rather than this one, but have you tried clearing all the system firmware settings back to defaults and then trying again? If that doesn't work please open a bug against fwupd and we can continue there.

emilas44, 2019-08-18:
Can you please post the link for the 'fwupd' project
I thought this was the same thing.

hughsie, 2019-08-18:
https://github.com/hughsie/fwupd/

emilas44, 2019-08-18:
Moved to https://github.com/hughsie/fwupd/issues/1292

closed 2019-08-18 by emilas44